Fedict/eid-mw #1: display.cpp:25:21: fatal error: dialogs.h: No such file or directory

joeyhub, 2015-05-03:
With --enable-dialogs=no I cannot compile at all.

joeyhub, 2015-05-05:
It compiles with:

    if NO_DIALOGS
    libbeidpkcs11_la_CXXFLAGS += -DNO_DIALOGS
    else
    libbeidpkcs11_la_CXXFLAGS += -I$(srcdir)/dialogs
    libbeidpkcs11_la_SOURCES += \
    	dialogs/dialogsgtk/dlgs_gtk.cpp \
    	dialogs/dialogsgtk/single_dialog.c
    endif

Changed to:

    if NO_DIALOGS
    libbeidpkcs11_la_CXXFLAGS += -DNO_DIALOGS
    endif

    libbeidpkcs11_la_CXXFLAGS += -I$(srcdir)/dialogs
    libbeidpkcs11_la_SOURCES += \
    	dialogs/dialogsgtk/dlgs_gtk.cpp \
    	dialogs/dialogsgtk/single_dialog.c

This is in version 4.1.2 from the beid website.

http://eid.belgium.be/en/binaries/eid-mw-4.1.2-v4.1.2.tar_tcm406-258906.gz

We also tried only including dialogs.h when ndef NO_DIALOGS and moving the namespace usage into the lower ndef check inside the function. However this causes a linked error as it turns out BadPin still produces a dialog (it is not preprocessed out with NO_DIALOGS).

This is a bit worrying that it cannot even build. For the official version on the website to be "half done" is a worrying state of affairs. Now I have it built but I worry about stability.

yoe, 2015-05-05:
This is some bit rot that occurred because compiling with --enable-dialogs=no isn't done as part of our continuous integration. I suppose I could look into that (shouldn't be too hard).

joeyhub, 2015-05-05:
It's a bit of whackamole with the missing NO_DIALOGS preprocessor directive. I added one here:

    bool CPkiCard::AskPinRetry(tPinOperation operation, const tPin & Pin,
													 unsigned long ulRemaining, const tPrivKey *pKey)
    {
    #ifdef NO_DIALOGS
    	return false;
    #else
    	bool bRetry = false;
    	bool bUsePinpad = m_poPinpad->UsePinpad(operation);

    	// Bad PIN: show a dialog to ask the user to try again
    	// PIN blocked: show a dialog to tell the user
    	if (!bUsePinpad)
    	{
    		DlgPinUsage usage = PinUsage2Dlg(Pin, pKey);
    		DlgRet dlgret = DlgBadPin(usage, utilStringWiden(Pin.csLabel).c_str(), ulRemaining);
    		if (0 != ulRemaining && DLG_RETRY == dlgret)
    			bRetry = true;
    	}
    	return bRetry;
    #endif
    }

joeyhub, 2015-05-05:
For using NO_DIALOG in object.c, in cardcomm/pkcs11/src/Makefile.am I had to add:
libbeidpkcs11_la_CPPFLAGS += -DNO_DIALOGS

Near to the same for CXX flags.

yoe, 2015-05-26:
Did you get this to work? If so, could you send a pull request? That way I don't have to do it again.

Thanks,

yoe, 2015-06-09:
This should be fixed now (at least in git). I've also added a check for this to our CI system, so it won't be repeated.

closed 2015-06-09 by yoe