Fedict/eid-mw #2: Design Flaw?

joeyhub, 2015-05-04:
This library provides a lot of inbuilt functionality which is nice for some but problematic for others.

I am not sure it makes sense to really build dialogs into lower level libraries then implant dummys (ie that always return yes, etc) when dialogs are off.

Nearly all the build problems I have are related to this.

I am not sure if the low level library hooks onto the application or pcscd but it would be more convenient for many of us if instead of integrating dialogs function hooks (if the lib is loads by the application) or some form of IPC were used.

The current architecture is problematic if people want to integrate this library into their own software. The integration it already provides could be more portable and detached. I think that the about, dialog and xpi items should really be separate programs all together. There is some isolation in the build system but it's still tricky if someone wants to turn these off completely. A separate package would enforce isolation and provide a better middleware product.

Frederikus, 2015-05-05:
Hello Joey,

This is indeed a point we have had quite a few discussions about in our
team in the past.

The reason we had to implement the dialogs in the low level pkcs11 library
was that we should provide the user with a notification whenever an
application is reading from his eID Card (by using our middleware). And
that the dialog message would be clear and simple (not to confuse the user
when the browser asks for a password when the user need to enter his PIN)

That being said, we will have a look at where we want to go with our
current buildsystem regarding the builds without dialogs.

Thank you for your remarks,
 Frederik

yoe, 2015-05-05:
For clarity: showing the dialogs isn't done because we like to annoy people, but because the privacy commission requires it (every time some application reads something from the card, the user must be informed). Given that requirement, this is the best possible option.

The whole point of the about tool is that it is built and shipped along with the middleware, so that it can show things like "what is the version of the currently installed middleware" by knowing so at compile time. Separating it into another package would defeat that purpose. However, I suppose we can make compiling it in optional.

joeyhub, 2015-05-05:
This all makes sense and actually for me the use case is very similar. I do want to be able to provide dialogs but they must be integrated into the application using it's own display logic as GTK is not appropriate for the device.

However perhaps I am interpreting the feature incorrectly. If default behaviour is to pass up responses to the caller anyway meaning that I can implement that functionality just using the externalised functions from the library, then this is the simplest way.

If this is the case then the problem goes back to simply having better support for an implementation of the NO_DIALOGS option.

yoe, 2015-05-06:
the -bin package only contains the dialog programs; these are "libexec" programs, not "bin" programs (the packages are only called -bin due to common packaging conventions).

If the plan is to improve integration with your environment, one approach could be to reimplement the dialogs in whatever user interface environment you're using; whether this is a sensible approach of course depends on your needs (that I don't know).

The intent for NO_DIALOGS is indeed that if you enable it, the library behaves a bit more like a standard PKCS#11 library. It only has to work ;-)

I'll try to fix things ASAP so that the NO_DIALOGS build works again, and will then integrate it with our continuous integration system (so that this won't repeat itself in the future).

Not a design flaw, so closing this issue (the other one on "things don't compile" will remain open).

closed 2015-05-06 by yoe