Fedict/eid-mw #4: Warnings when building with VS2015

dheijl, 2015-10-30:
minidriver:

> object.c: c:\users\dh\documents\visual studio 2015\projects\eid-mw\cardcomm\pkcs11\src\object.c(492): warning C4244: 'function': conversion from 'CK_ULONG' to 'CK_BYTE', possible loss of data

pkcs11:

> logbase.cpp: c:\users\dh\documents\visual studio 2015\projects\eid-mw\cardcomm\pkcs11\src\common\logbase.cpp(844): warning C4477: 'fprintf_s' : format string '%s' requires an argument of type 'char *', but variadic argument 3 has type 'const wchar_t *'
c:\users\dh\documents\visual studio 2015\projects\eid-mw\cardcomm\pkcs11\src\common\logbase.cpp(844): note: consider using '%ls' in the format string
 c:\users\dh\documents\visual studio 2015\projects\eid-mw\cardcomm\pkcs11\src\common\logbase.cpp(844): note: consider using '%lls' in the format string
 c:\users\dh\documents\visual studio 2015\projects\eid-mw\cardcomm\pkcs11\src\common\logbase.cpp(844): note: consider using '%Ls' in the format string
 c:\users\dh\documents\visual studio 2015\projects\eid-mw\cardcomm\pkcs11\src\common\logbase.cpp(844): note: consider using '%ws' in the format string

yoe, 2015-11-01:
First has been fixed now. Will look at the others at a later time

yoe, 2015-12-22:
Above commit should fix those warnings.

closed 2015-12-22 by yoe