Fedict/eid-mw #8: eid support on Chrom(e|ium) Linux and MS Edge fails

wget, 2015-12-06:
Hope that question hasn't been already asked on your Google Group. Sorry if that is the case.

Following a recent discussion with @roidelapluie on Twitter:
https://twitter.com/wget42/status/661175349204688896
and with Wouter Verhelst:
https://twitter.com/wouter_verhelst/status/673515683842564097

I don't know what should be the current situation with the eID middleware. But here are the tests I performed.

# Windows

> Windows 10 x64 Pro 1511, using quick install

* Microsoft Internet Explorer: working
[image]

* Google Chrome: working
[image]

* Mozilla Firefox: working in all use cases
[image]

  - If using the extension
     - need to use eID middleware, 

  - If not using the extension
     - need to go to `Preferences > Advanced > Certificates > Security Devices > Load` and specify the file `beidpkcs11.dll` in the field `Module filename:`. We need to take the version located either in `system32` or `SysWOW64` depending on the version of Firefox used (32 bits or 64 bits).
  src.: http://faq.eid.belgium.be/fr/utilisation.html section about Adobe Acrobat

* Microsoft Edge: not working.

  The authentication certificate and card reader are recognized but the authentication step fails.
[image]
[image]
[image]
[image]

# OS X:

# GNU/Linux:

> Arch Linux x64, rolling release

* Mozilla Firefox: working in all use cases.

  - If not using the extension,

     - if using the middleware, need to specify the file `beidpkcs11.dll`.

     - if not using the middleware, need to use the package `opensc` and locate the file `/usr/lib/opensc-pkcs11.so`.
       src.: https://roidelapluie.be/tax-on-web.html

  - If using the extension, need to use eID middleware, 

* Chrome/Chromium: not working.

  - If specifying `/usr/lib/opensc-pkcs11.so` from `opensc`
  
  - or using the lib `beidpkcs11.so` from the eID middleware whose path is specified using `about-eid-mw`.
  
  In all use cases, the browser detects the authentication PKCS11 certificate but we got the error
  `ERR_BAD_SSL_CLIENT_AUTH_CERT`
  
  When running from CLI, the error, we have the following error:
  `[6153:6177:1206/125655:ERROR:ssl_client_socket_openssl.cc(1056)] handshake failed; returned 0, SSL error code 1, net_error -117`

  openssl version: core/openssl 1.0.2.e-1
  Mozilla Network Security Services version: core/nss 3.20.1-1
  Chromium version: extra/chromium 47.0.2526.73-1

  src.: https://github.com/Fedict/eid-mw/wiki/ChromeLinux

  I am not the only one having the issue: http://gathering.tweakers.net/forum/list_messages/1653910

  According to the answers I found on the net, it is asked to double check if date is correct on the system. I checked myself and ntp is running flawlessy. Even inside the browser, the date is correct when checked in Javascript:
  `var date = new Date(); date.getSeconds();`
  
  If have found other answers asking to compile Chromium with BoringSSL instead and not with openssl. But as I'm an Arch Linux user and I know most (if not all) changes brought to BoringSSL ends up being integrated to OpenSSL, I assume the version I have (core/openssl 1.0.2.e-1) hasn't issues.

yoe, 2015-12-08:
I'm sorry, but this isn't really an actionable bug report.

The short answer to "which browsers are supported" is simply, "all of them, as far as technically possible". That is, if it is possible to persuade a browser to support the eID, then we will do our best to make it work. However, the ability to support a smartcard for a particular browser depends on the browser's developers providing ways to make that possible; some choose not to.

As to the browsers you mark as failing:

* We've just re-tested Microsoft Edge ourselves, and that works for us. For help in getting it to work properly for you as well, please contact the Fedict servicedesk.
* Google Chrome for Linux used to work, but that is indeed no longer the case. This is due to changes made in Google Chrome, not due to changes made in the middleware. If we find the root cause, we might be able to reinstate support for Google Chrome; but since the Chrome developers explicitly do not support smart card usage on Linux, this may simply be a conscious choice on their part, in which case there's nothing we can do.
* I'll also note that the firefox extension does two things; the first (and the whole point of it, its *raison d'être*) is to locate and activate the PKCS#11 module; the second is to add a custom icon for the eID card, so that if Firefox shows a browser dialog related to your eID, it uses the eID icon. As such, not loading the extension does indeed mean you'll have to manually load the PKCS#11 shared library, but that's to be expected.

Finally, while there is indeed support for the eID in OpenSC, please note that Fedict does not support this third-party software.

Since there's not really something we can do about the issues you mention, I'm going to close this bug report.

wget, 2015-12-08:
Thanks for your feedback.

* Google Chrome/Chromium on Linux: 
  > but since the Chrome developers explicitly do not support smart card usage on Linux

  Do you have any source about this? ^

  This might be of interest: https://code.google.com/p/chromium/issues/detail?id=42073

* MS Edge:
Which version of Windows 10 (file manager -> file -> about, with build number) and Edge do you use? The Windows 10 version I tested was a 1511 I installed from scratch (not an update from 10240) with the iso available (`fr_windows_10_multiple_editions_version_1511_x64_dvd_7223728.iso`) on my MSDN account.

The reader I have is an Omnikey 3021. Do you think websites for the Belgian eID could work with the Digipass 870 provided by Belfius? Belfius is actually using that reader with the Belgian eID to update our personal data on their platform.

Why aren't you relying on OpenSC completely? What are the differences between Fedict eID pkcs11 lib and the one from OpenSC?

yoe, 2015-12-08:
Older versions of the official middleware were based on OpenSC, but the current version is not. This choice was made several years ago; I can't comment on why that is the case. It still implements the same API that is defined in the PKCS#11 standard, but beyond that the code is not related.

The digipass 870 should work; we've done various tests with that device ourselves.

I have no source for the claim about chrome. I might be mistaken. At any rate, I've not given up, but for now it's not possible to get it to work.

Please note though that the github issue tracker is not a support channel for getting basic functionality to work. If there's a bug in the code, then filing an issue here is certainly appropriate. However, that does not appear to be the case. Again, if you want help with getting Microsoft Edge to work, please contact the Fedict servicedesk who will gladly help you get things working.

yoe, 2016-02-17:
I may have been a bit quick closing this issue. It turns out that at the time you reported these issues, some components on our testing machines may have been a bit behind the times. At any rate, we can reproduce them now (even though we couldn't originally).

I found out what the root cause was of the Chrome issue (which turns out to be something else than what I originally thought). I've opened #9, which has a link to the relevant issue in the Google Chrome bugtracker. The bug in Chrome was introduced when the Chrome developers switched to BoringSSL as their TLS implementation; they've acknowledged that, and are working on a fix.

I'll reopen this issue to track the Microsoft Edge case. We're not exactly sure what's going on there, yet, but it appears there's some change in Edge which makes it lose its connection with the smartcard.

Ioannes1, 2016-03-29:
Hello yoe,
I am also using Windows 10 ,MS Edge and the Digipass 905.
The EiD-reader is recognized but the systems fails at the authentication ( screens are exactly looking like as described by wget above).
Is there already a solution for this ? What are the different alternative options?
kind rgds

yoe, 2016-03-29:
@Ioannes1 we are working with Microsoft on the problem. Other browsers under Windows 10 don't have issues; for now, I recommend you try Internet Explorer, Chrome, or Firefox.

When we have found (and fixed) the issue, this bug will be closed.

Frederikus, 2017-01-10:
Edge support has been added to the 64-bit minidriver by removing the windows dependancy libraries that we used to display dialogs; i.e. the dialog to notify the user that he needs to enter his PIN on his secure PIN pad reader.
This because Edge is a universal app., and it prohibits the use of some older win32 libraries.

This change will also influence other webbrowsers that use the 64-bit minidriver. (They will also no longer show the notification to the user that his secure PIN pad reader is waiting for a PIN entry)

closed 2017-01-10 by Frederikus