Fedict/eid-mw #9: Google Chrome/Linux

yoe, 2016-01-20:
Currently, eID authentication on Google Chrome (or Chromium) for Linux does not work. This seems to be due to a change in Google Chrome made after Chrome 46.

Filed a bug report at https://code.google.com/p/chromium/issues/detail?id=579480

dagwieers, 2016-06-07:
This issue is still valid for Chromium v51 (51.0.2704.79).

simonh1000, 2016-06-22:
Any update as it is tax-paying time....!

See below for Chrome Version 50.0.2661.26 beta (64-bit) on Ubuntu 16.04 

[image]

Frederikus, 2016-06-22:
Hello Simon,

We checked recently with the people of FAS, and they are working on it. (being able to set up a secure connection when the client only sends a leaf certificate).
Would using firefox be an acceptable workaround to fill in your taxes in the meantime?

I'll send them a reminder and point them to your post.

Wkr,
 Frederik

wget, 2016-09-02:
As soon Chrom(e|ium) 54 is released, we will be able to mark this bug as resolved \o/

https://bugs.chromium.org/p/chromium/issues/detail?id=579480#c29

yoe, 2016-10-25:
Chrome 54 has been released now.

Additionally, since the same problem now occurs with Safari 10 on macOS 10.12, it is reasonable to say that this is no longer a case of a single browser not working. Therefore, a change has been made to the federal authentication service (FAS) to work with browsers that exhibit this behaviour.

The two combine to resolve this bug, so closing the issue.

closed 2016-10-25 by yoe

dagwieers, 2016-10-25:
Users rejoice !