Fedict/eid-mw #13: Not working on OSX

simonh1000, 2016-07-07:
I've now tried with Chrome and Firefox and `Your electronic identity card (eId) could not be verified.` while on Safari I don't event get that far before it hangs. I'm using an HID OMNIKEY 1021 reader and the java card reader software can read my E+card.

Is there a guaranteed route to getting connected - I lose hours each year trying to complete my legal obligations and feel that the State should do more to make things easier to use.

Frederikus, 2016-07-08:
Hello,
For operational issues, please contact our servicedesk:
http://eid.belgium.be/en/contact/contactform

For firefox, you will also need an addon that tells firefox where to find our pkcs11 module.
You can find this addon here:
https://addons.mozilla.org/nl/firefox/addon/belgium-eid/
(make sure it is enabled, and that you are using firefox 47)

yoe, 2016-07-15:
Since this does not expose an actual bug in the eID software (or if it does, the issue does not contain enough information for us to debug and fix it), I'm going to close this issue.

Feel free to contact our support people (Frederik gave the link in his comment) for help getting your eID setup to work. If they find there's a bug, they'll certainly let us know and we'll fix it.

closed 2016-07-15 by yoe