Fedict/eid-mw #16: PPA installs outdated Firefox extension in /usr/share/mozilla/extensions

robvdw2, 2016-07-16:
Not a big problem, but I had to fix several installs, so other people might have the same issue. 

If you follow the instructions on http://eid.belgium.be/en/using_your_eid/installing_the_eid_software/linux on Ubuntu, you will end up with an outdated Firefox extension that does not update automatically. 

Steps to reproduce:
 - Clean Ubuntu 16.04 install from ISO (I used Ubuntu MATE 16.04 in a KVM/QEMU VM)
 - Update packages (Firefox version is 47.0)
 - Install the PPA by installing the eid-archive.deb package from http://eid.belgium.be/en/using_your_eid/installing_the_eid_software/linux
 - apt-get update && apt-get install eid-mw
 - Restart Firefox
 - _"The configuration of the Belgian electronic identity card has failed. Is the eID Middleware installed?"_
 - The version installed in /usr/share/mozilla/extensions is 1.0.18.1-signed

The solution is relatively straight forward, but not obvious to all users and not included in the main set of instructions on the website:
 - Go to https://addons.mozilla.org/nl/firefox/addon/belgium-eid/ and install the add-on from there (version 1.0.21 will be installed into your home .mozilla directory, which will override the old version that stays in /usr/share/mozilla)
 - Restart Firefox & problem solved

yoe, 2016-07-20:
Right, sorry, forgot about that. This is a problem from FF 47 onwards, which was released just before I went on holiday. I was going to release updated packages which contain the newer plugin, but haven't done so yet.

Will deal with this today.

yoe, 2016-07-20:
Actually, I did *not* forget about that, but apparently made a mistake so that the new version of the add-on which was *supposed* to be shipped with 4.0.18 is in fact not shipped.

Regardless, we'll fix this :-)

yoe, 2016-08-23:
Just pushed something to the v4.1 branch (i.e., the "candidate" package repositories) which should (finally) fix this.

I was a bit confused with my original statements on this issue. The problem is that there are two versions of the plugin: one which works with firefox 46 (and below) and one which works with firefox 47 (and above). There was an API change, and making sure this continues to work with both versions is complicated.

I didn't immediately update, since at the time ff 47 wasn't everywhere yet, mostly due to ff 45, which is an ESR release. My recent change pushes the addon for ff 47 where the distribution uses that (hopefully). If my tests reveal no issues, I'll release that in the next few days.

yoe, 2017-01-05:
Current versions of the beid-mozilla-extension package install version 1.0.22 of the plugin, which works with all versions of firefox (i.e., the ESR as well as the normal release). This issue should be fixed now.

closed 2017-01-05 by yoe