Fedict/eid-mw #20: Support Fedora 25?

swertz, 2016-12-30:
Hello,

Fedora 25's been out for a while now, but when installing the RPM to configure the repositories to fetch the eid software, it doesn't work.

I had to manually hack the repo's config file to fetch the RPMs for Fedora 24. I could install them, but there is still an issue: the test page fails on Firefox ("SSL_ERROR_HANDSHAKE_FAILURE_ALERT"), and the eid-viewer does not recognise my card reader (works fine on ubuntu).

Any chance this could be looked at?

swertz, 2016-12-30:
Update: restarting the pcsc service did the trick, now it works. So you'll only need to update the repository for 25 (and I just noticed I might be on the wrong github repo for that particular issue)...

yoe, 2016-12-30:
This is the right repository. I didn't notice Fedora 25 had been released, so I didn't build the packages yet. I'm on holiday now due to the end of year, but will build& release them as soon as I get back.

swertz, 2016-12-30:
Great, thanks! I've compiled it myself in the end but I still get the above problem though...

yoe, 2017-01-04:
I had the same issue when I first installed the packages that I'm about to release on a VM. However, after a reboot of my (virtual) system, the problem resolved itself. I haven't been able to reproduce it since.

Can you still reproduce the problem? If not, I'll close this issue after the packages are released.

yoe, 2017-01-04:
Packages released now. Waiting for feedback.

yoe, 2017-01-09:
No feedback received, assuming all is well. If not, please reopen the issue with details of your setup.

closed 2017-01-09 by yoe

swertz, 2017-01-09:
Sorry, I couldn't try the update until tonight... I'll take a look, thanks for looking into it!

swertz, 2017-01-11:
Only rebooting did not work, but I have reinstalled it using the version in the repos thanks to your release and now it works! Many thanks for your quick help, and sorry again for not getting back to you.