Fedict/eid-mw #23: Hello world with Visual Studio 2015 not working.

shamun, 2017-02-07:
I just git clone and then using Visual Studio 2015 did build/compile. On first attempt it failed. Then i did some scripts execute after that the build was success.

But when i now go to test folder i dont see any sample. Is there any sample to make sure my build / compile working?

shamun, 2017-02-07:
Build failed:

[image]

Even the DLL is in reference:

[image]

shamun, 2017-02-07:
Retried but still same.

[image]

[image]

shamun, 2017-02-07:
C code worked but not all, C# sample failed badly. Please update so that we can clone and retest.

[image]

Frederikus, 2017-02-09:
Hello Shamun,

Thank you for your feedback, we will remove the app.config file.

What is the path of the net.pkcs11 library that is being referenced in your
project?
Could it be that you are trying to link the 64-bit version of
net.pkcs11.dll into a 32-bit application?

Wkr,
 Frederik

Frederikus, 2017-02-09:
Looking at your second screenshot, I see its the 32-bit version and your
project is any-cpu, so on a 64-bit system, you will not be able to load the
32-bit version of the net.pkcs11.dll

shamun, 2017-02-17:
I was unable to resolve it with C#. it keeps failing in my end. but the C code is working, so i will be using the C code.

closed 2017-03-01 by yoe