Fedict/eid-mw #24: Issue writing the picture to file

JonathanGailliez, 2017-02-14:
Hello,

I'm new to the SDK and in order to learn, I'm extending the C example you provide in the project (eid-mw/doc/sdk/examples/C/) in order to write the photo contained in the card in a JPG file. When I execute the program on my mac I get a proper image that I can open and work with. When I run the same project on Windows I get a corrupted JPG file that I can't open. I attached the 2 files below.

Here is the code I've added to the example provided:
    
    char *photoPath = "/tmp/eid.jpg";
    FILE *photoFile = fopen(photoPath, "w");
    CK_CHAR_PTR pFilename = (CK_CHAR_PTR)"PHOTO_FILE";
    CK_VOID_PTR pFileValue = NULL;
    CK_ULONG FileValueLen;
    CK_RV retVal = Beidsdk_GetObjectValue(pFunctions, session_handle, pFilename, &pFileValue, &FileValueLen);
    if(retVal == CKR_OK)
    {
        unsigned long counter = 0;
        if(pFileValue != NULL)
        {
            //fwrite(pFileValue, 1, FileValueLen, photoFile); // Give the same corrupted file
            while (counter < FileValueLen)
            {
                fprintf(photoFile, "%c", *(pFileValue+counter));
                counter++;
            }
        }
    }
    else
        printf("error 0x%.8x Beidsdk_GetObjectValue\n",retVal);

What can be the cause of the fact that when I write the file on Windows it is corrupted and not when I do it on OS X? Do you need further information?

Thank you for your time.

[image]
[image]

Frederikus, 2017-02-14:
Hello Jonathan,

Can you try to use fopen with "wb" , see
https://msdn.microsoft.com/en-us/library/yeby3zcb.aspx :
"b
Open in binary (untranslated) mode; translations involving carriage-return
and linefeed characters are suppressed."

Wkr,
 Frederik

JonathanGailliez, 2017-02-14:
Hello Frederik,

it works using fopen with the wb option.

Thanks a lot for your help. I'll close the issue.

Best regards,
Jonathan.

closed 2017-02-14 by JonathanGailliez