Fedict/eid-mw #25: How to make the GUI disappear?

shamun, 2017-02-17:
I have a Robot, which have USB id card reader connected on it. With sensor it talks and walks, finally when it detect human it says please insert your ID card reader, to read your record, when you are done please remove your eID card from the reader.

Now in the Robot there is no MONITOR, no KEYBOARD, no MOUSE. how do i tell eid-mw framework to please disable the DIALOG window?

[image]

this is the code i am using: https://github.com/Fedict/eid-mw/blob/master/doc/sdk/examples/C/src/data.c#L167

```
/* Read exactly one value from the card. */
CK_RV Beidsdk_GetObjectValue(CK_FUNCTION_LIST_PTR pFunctions, CK_SESSION_HANDLE session_handle,CK_CHAR_PTR pName,CK_VOID_PTR *ppValue, CK_ULONG_PTR pvalueLen)
{
	CK_RV retVal = CKR_OK;
	CK_ULONG data = CKO_DATA;
	CK_ATTRIBUTE searchtemplate[2] = {{CKA_CLASS,&data,sizeof(CK_ULONG)},{CKA_LABEL,(CK_VOID_PTR)pName,(CK_ULONG)strlen(pName)}};
	*pvalueLen = 0;

	//initialize the search for the objects with label <filename>
	retVal = (pFunctions->C_FindObjectsInit)(session_handle, searchtemplate, 2); 
	if (retVal != CKR_OK) { return retVal; }

	CK_OBJECT_HANDLE hObject = 0;
	CK_ULONG ulObjectCount = 0;
	//find the first object with label <filename>
	retVal = (pFunctions->C_FindObjects)(session_handle, &hObject,1,&ulObjectCount); 
	if (ulObjectCount != 1) { retVal = CKR_GENERAL_ERROR; }
	if (retVal != CKR_OK) { goto finalize; }

	/* NULL_PTR as second argument, so the length of value is filled in to
	 * retValueLen. See the definition of C_GetAttributeValue in the PKCS#11
	 * standard for more details.
	 */
	CK_ATTRIBUTE attr_templ[1] = {CKA_VALUE,NULL_PTR,0};

	//retrieve the length of the data from the object
	retVal = (pFunctions->C_GetAttributeValue)(session_handle,hObject,attr_templ,1);
	if(retVal != CKR_OK || ((CK_LONG)(attr_templ[0].ulValueLen) == -1)) { goto finalize; }

	*ppValue = malloc (attr_templ[0].ulValueLen);
	if (*ppValue == NULL) { retVal = CKR_GENERAL_ERROR; goto finalize; }

	attr_templ[0].pValue = *ppValue;
	/* now run C_GetAttributeValue a second time to actually retrieve the
	 * data from the object
	 */
	retVal = (pFunctions->C_GetAttributeValue)(session_handle,hObject,attr_templ,1);
	*pvalueLen = attr_templ[0].ulValueLen;

finalize:
	//finalize the search
	retVal = (pFunctions->C_FindObjectsFinal)(session_handle); 
	return retVal;
}

```

Frederikus, 2017-02-17:
> mw/blob/master/doc/sdk/examples/C/src/data.c#L167

hello Shamun,

we were given permission to remove this dialog recently.
As there are other means of informing the card owner that his card will be
read out.

We have a test release available which no longer shows this dialog:
http://eid.belgium.be/en/using_your_eid/installing_the_eid_software/windows
version 4.2.1

Wkr,
 Frederik

shamun, 2017-02-18:
Dear Frederik,

This is GREAT!!! very good news. Finally we can use it in robotic. 
Thank you very much, i can now make my progress much easier way.

Appreciate your support and such a great work.

Regards
/Sha

shamun, 2017-08-21:
@Frederikus:

- 21 Aug 2017 i have the same problem. I tried to installed 'http://eid.belgium.be/en/using_your_eid/installing_the_eid_software/windows` but link is dead?

[image]

[image]

I need to kill that GUI. Last time it worked by using version 4.2.1 but now that URL is not alive. Can you please advise?

shamun, 2017-08-21:
In Feb/2017 i used following version. But i cant find it anymore by following your previous URL. 

[image]

yoe, 2017-08-22:
Every version of the eID middleware since 4.2.1 has the dialogs disabled. You can just choose the most recent version; we did not create a specific version with the dialogs disabled, or anything of the sorts.

closed 2017-08-22 by yoe