Fedict/eid-mw #27: BUG - even ID card reader is connected i keep getting the status "reader_unplugged"

shamun, 2017-03-08:
Please check this code: 
=

https://github.com/shamun/eid-mw/blob/828ccf2379db638c4b2c066db9b9390d6c572fd2/doc/sdk/examples/C/src/data_more.c

To build, replace the  existing `data.c` with my code.

Use flow:
=

- Kiosk running the `data_more.c` with 2 second interval
- when ID card is inserted, data comes, when ID card is invalidly inserted shows "invalid_insert", when ID card reader is removed from the PC shows "reader_unplugged" or on invalid insert

Problem:
=

- even the ID card reader is connected to the PC
- even the drivers are all upto date
- even the code has no problem

Time to time while scanning i see "reader_unplugged" and it just keep showing. Which breaks the whole application and confuse everyone

Temporary solution:
=

- after  a while if you remove the USB ID card reader and reconnect the USB manually to the same port or another port, then by crossing finger you can see the card reader start to work

This problem i have been suffering for more then 2 year almost. yet its happening. Could you guys please kindly check on this issue? I have checked almost all the related cause but its absolutely abnormal when it shows "reader_unplugged". cause card reader is connected and drivers are installed. but still "reader_unplugged" keep showing.

Please advise if you have any suggestion.

Best reg
/Sham

Frederikus, 2017-03-08:
Hello Shamun,

We are aware that some card readers do not awake anymore after entering sleep mode.
Can you check if this is the case with your issue?
Should your reader have a power indication LED, can you confirm it is still on when you are experiencing the issue?

Wkr,
 Frederik

shamun, 2017-03-08:
Dear Frederik,

"1) We are aware that some card readers do not awake anymore after entering sleep mode.
Can you check if this is the case with your issue?
2) Should your reader have a power indication LED, can you confirm it is still on when you are experiencing the issue?"

1
= 

- Glad you know this issue
- I have 5 unique USB 2.0 ID Card readers, in past there was one which goes to sleep mode and never comes back until you remove the USB 2.0 and reconnect it. But out of 5 it was only that 1 which had that issue. Since i realise that it was some issue, then i more focused on finding something more stable then i found one "HID Omnikey 3121" it has official most recent windows 10 driver
- we removed old ID card readers and i have recently i think 1 week ago connected the HID Omnikey 3121 onsite (For live customer kiosk)

2
=

- Since yesterday with all latest code, all new updates still in whole day i have now atleast 1 time this same issue
- when the issue happen, i have instantly remote access onsite to investigate it. What i found is that the ID card reader was on, card reader is connected to the USB 2.0, my code was running but for 10 to 20 seconds i was keep getting "reader_unplugged" alert in my mobile. And then automatically after 20 seconds it was again back to normal.

Summary:
=

I confirm, the reader is connected to the USB 2.0, i confirm the ID card reader was getting 5 volt constant power supply to keep it alive, i confirm there was no phyiscal issues with PC to the reader relation. 

My wild guess 
=

are we not doing something wrong in drivers level or so? 

(cause its being while i had this issue, and i have tried so many random ID card readers, but still i am experiencing this same problem, i have tried myself constantly weeks after weeks with infinite loops after loops, but on live with real customer on kiosk mode this problem occures, which means random ID cards might be incompatible??? which triggers the ID card reader itself into sleep mode for a few seconds and if the ID card reader is smart it auto start back again??? and that is the stage i was getting that "reader_unplugged" alerts???)

reg
/sham

yoe, 2017-03-08:
Glancing over your code, I see the following:

	retval = beidsdk_GetData();

	if (retval == CKR_OK) {
		//printf("OK");
	}
	else {
		printf("reader_unplugged");
	}

This is pure stupidity. There are loads and loads of possible return values, and they really do not all mean "the reader was unplugged". Please check [the PKCS#11 documentation](https://github.com/Fedict/eid-mw/blob/master/doc/sdk/documentation/pkcs-11v2-11r1.pdf), section 11.1, for a list of all possible return values and their meaning.

Some of these errors are recoverable by fixing your call; some of these errors are recoverable by reinitialising the PKCS#11 library (i.e., `C_Finalize()` followed by `C_Initialize()` followed by whatever you were trying to do again); some of them are not recoverable; and yes, `CKR_DEVICE_REMOVED` *may* mean that the reader was unplugged. By just claiming that every possible return value that is not `CKR_OK` is a "reader was unplugged" condition, however, you're muddling the waters for everyone.

Closing this issue, since it very obviously is not a bug in the middleware but a bug in your application code.

closed 2017-03-08 by yoe

shamun, 2017-03-08:
OK - Thank you, i will work on my existing version based on that PDF to have more detail overview of the return state from middleware.