Fedict/eid-mw #32: Is it possible to make the eID viewer get focus upon inserting a card?

Wannes123, 2017-06-20:
Hi,

Excuse me in advance if this is not the appropriate place for a question/suggestion.

Is it possible to make the eID viewer get focus (pop up to front) upon inserting a card?

This would be very useful when reading a lot of cards as part of a daily routine.

Thanks for considering,
Wannes Schols

Frederikus, 2017-06-27:
Hello Wannes,

We've discussed this and are not convinced this behaviour that would add much value to the viewer. 

Applications that would suddenly jump to the front sounds like rather odd behaviour, that can be nasty depending on what the user was doing.

Besides that, I'm also not sure if its possible to grab the focus from another application, but haven't really looked into that yet. 

But thanks for bringing up idea's, we surely welcome that

Wannes123, 2017-06-27:
Frederikus,

We think the user would not be interrupted since he is inserting an eID, so he knows that the eID will get focus. An option to enable/disable perhaps? :)

Grts,
Wannes

yoe, 2017-06-27:
- There are many ways in which the eID may be used. The fact that you're inserting an eID does not *always* mean you want to use it with the viewer, even if that viewer is running currently. 
- Bringing something to front/in focus isn't always easily doable, and can certainly not be done in a cross-platform way (e.g., X11 (as used on Linux, amongst others) does not allow this, nor does macOS; we're not certain about Windows)
- The person who inserts the eID card need not necessarily be the person who is using the computer. I would be annoyed if an eID viewer pops to front while I'm doing something else, because my customer/client/whatever's card was inserted prematurely.

As such, bringing an application to the front just because an eID card was inserted isn't something we'll implement ourself, but a patch that allows it (while disabling it by default) it might be accepted.

closed 2017-06-27 by yoe