Fedict/eid-mw #33: eID Viewer beeps when reading a card on macOS

Dev1an, 2017-06-21:
When I insert a card and the eID Viewer app is open, it beeps. The beep I'm referring to is the the standard warning/error sound on macOS (called funk), but when I go and read the log, there are no errors.

So I think the sound should either be changed (to Glass for example) or removed.

yoe, 2017-06-22:
Now that you mention it, yeah. I hadn't noticed before, because the mac that I do my development on usually has sound disabled.

I'm not sure where the sound is coming from; it's certainly not something I've added myself. Best guess is that when we [show the sheet](https://github.com/Fedict/eid-mw/blob/master/plugins_tools/eid-viewer/OSX/eID%20Viewer/eID%20Viewer/AppDelegate.m#L224), macOS does this all by itself.

This would seem to be supported by the fact that the sound occurs twice, and we do have two sub-states that happen while the sheet is shown (one for reading identity data, one for reading certificates).

Not sure how to fix that. Suggestions welcome :)

yoe, 2017-07-05:
Thanks to https://stackoverflow.com/questions/44921993/create-a-sheet-that-isnt-a-warning-message :-)

closed 2017-07-05 by yoe