Fedict/eid-mw #35: Log level setting not persisted in eID viewer

Wannes123, 2017-06-22:
Hi, 

The setting for the log level in the eID viewer is not persisted upon exit/start of the program.

Grts,

Frederikus, 2017-06-22:
Hello,

That is correct, if there is a need for it we can surely add it.

Wkr,
 Frederik

Wannes123, 2017-06-22:
Hi,

We are going to use the eID viewer in a customer facing scenario where it will be very frequently used. So it would be appreciated if this was implemented.

Grts,

Frederikus, 2017-06-27:
Just added this by storing the log_level in the windows registry

closed 2017-06-27 by Frederikus

Wannes123, 2017-06-27:
Hi,

That is great, but we were expecting it to be added to the settings file:

C:\Users\<user>\AppData\Roaming\eIDViewer\eID_Viewer.exe_Url_ebwncvpfrkyvuzi3d3p35zgvfmsietz0\4.2.8.3252

In this file, the value for the setting AlwaysValidate is also kept. As we want to rollout the eID viewer corporate wide, and change both AlwaysValidate and the LogLevel, it would have bene nice if we only had to change the settings file.

But thanks anway :)

Frederikus, 2017-06-27:
Aha,

Indeed that AlwaysValidate setting is stored by using .net Properties.Settings.
Wasn't actually intended to have that changed by default.

The logging setting was placed in the registry, next to the middleware logging config setting (only on HKCU for now) so it would be easier accessible from outside the application :)