Fedict/eid-mw #36: Is it possible to bypass all certification validation in eID viewer?

Wannes123, 2017-06-22:
Hi,

Currently it takes about 15 seconds before you can drag-and-drop data from the eID viewer:
+- 10 seconds before data is visible
+- 5 seconds for the loading bar handling certificates etc. to complete.

Is it possible to skip all certificate related processing, thereby speeding up the process of drag-and-drop by +- 5 secs?
Or, alternatively, is it possible to allow drag-and-drop form the moment the eID data is visually presented? (Currently it crashes when doing a drag-and-drop the moment the data is visualized, as mentioned in bug #31)

Thanks,

yoe, 2017-06-22:
Yes and no.

You can disable certificate *verification*. Just head to the "Certificates" tab and disable the "Validate always" checkbox. This stops the viewer from performing a number of OCSP requests and validating that the certificates are valid.

You can *not* disable the *reading* of the certificates. The XML files are supposed to contain those too, so before any drag-and-drop operation can be started, they must have been read out.

closed 2017-06-22 by yoe

Wannes123, 2017-06-22:
OK, thank you!

Wannes123, 2017-06-27:
Hi,

If I may revisit this topic.

Would it be possible to add an option 'Include certificates in drag-and-drop', and if this is disabled, you can do a drag-and-drop before the certificates are read in (thereby speeding up the wait time between inserting the eID and doing a drag-and-drop)?

Thanks for considering,

yoe, 2017-06-28:
If we go down that route, then the next question will be to drop the photo, and then the next question will be to also drop the address file, and then can we please just only drop the first name, and then people start complaining because they reconfigured things earlier and they forgot and now it's not working anymore and please can I have my pony back because AAAHhh.

This is not meant as an "I only need one item from the card" API. If you need that, there is an [official SDK](https://github.com/Fedict/eid-mw/tree/master/doc/sdk/documentation) which does allow that. Instead, this is meant as a simple way to provide some basic functionality to some environments where reading from the card directly would be either impossible or overkill. That it is slow is known, but we don't think it's necessary or useful to change that.

Wannes123, 2017-06-29:
Ok, I understand.

Before I delve deeper in this, can you please answer the following?
- The new SDK is written in C?
- The new SDK is the only way to access new eID cards (post-2014) from a custom build application/website? Yuu cannot use Java/Java applets?
- Any hints on how one would go about accessing  eID card information from Javascript? You would call a C function from your Javascript?

Thanks for your input,

Frederikus, 2017-06-29:
Hello Wannes,

regarding your points:

   - The new SDK is written in C?

-) The new SDK (pkcs#11 lib) is a C Library (  written in C and C++), we
also distribute a C# wrapper of the pkcs#11 library

   - The new SDK is the only way to access new eID cards (post-2014) from a
   custom build application/website? Yuu cannot use Java/Java applets?

-) No, there are many ways to read out eID cards, but the ones we support
are this SDK (pkcs#11 lib), as well as the backend of the eID Viewer (which
can be used against your own application)
-) There are several third party pkcs#11 wrappers available, should you
prefer writing in other languages
-) You should be able to use our pkcs#11 library from withing java,
see:
http://docs.oracle.com/javase/8/docs/technotes/guides/security/p11guide.html#P11Provider

-) We do no longer maintain our own Java applet, but third party vendors
might

   - Any hints on how one would go about accessing eID card information
   from Javascript? You would call a C function from your Javascript?

-) A browser will not give some javascript code from a website acces to our
pkcs11 library, that wouldnt be safe.
-) I'm not really into webapps, but you might want to have a look at java
webstart applications

Wkr,
 Frederik