Fedict/eid-mw #37: Problem with xml generated by drag-and-drop

fluchette, 2017-06-26:
Hello,

we have a problem with the xml generated by eid-viewer (4.2.8) drag-and drop.

If there are several first names, example : Antoine Laurent H

XML generated :

<firstname>Antoine Laurent</firstname>
    <middlenames>H</middlenames>

instead of

<firstname>Antoine</firstname>
    <middlenames>Laurent H</middlenames>

Thanks

yoe, 2017-06-27:
This is normal. Please see the [documentation](https://github.com/Fedict/eid-mw/blob/master/doc/sdk/documentation/belgian_electronic_identity_card_content_v2.8.a.pdf) of the data as it is stored on the card.

There are two fields for "given names": the first is the field with tag 8, which is marked as "First two given names"; the second is the field with tag 9, which is marked as "First letter of 3rd given names".

The field in the XML file may have been chosen suboptimally; after all, it does not contain the card holder's "middle names", but in fact contains the third name's first letter. However, for backwards compatibility, we're not going to change that.

closed 2017-06-27 by yoe

fluchette, 2017-06-27:
Ok but the tag is called 'firstname'.

Why to inform the second first name in this tag ???

yoe, 2017-06-27:
Yes, it is. As I said, that was probably a mistake, but it's too late to fix that now.