Fedict/eid-mw #39: Warning pop up

armandomusto, 2017-07-07:
Hi,
I integrated the eID Viewer into my .NET Desktop application, but when I initialize the reader this warnings pop ups.  How is it possible to hide it?

Thanks.

[image]

yoe, 2017-07-07:
Upgrade to middleware v4.2; in that version, these messages have been removed.

closed 2017-07-07 by yoe

armandomusto, 2017-07-07:
Hi joy,
thanks for the fast answer.
Which dll do I need to update?

yoe, 2017-07-07:
The update was done for the PKCS#11 module (i.e., beidpkcs11.dll), but don't just update that DLL by itself; it might interact badly with the other parts of the middleware that are already on your system. Instead, you should just install the newest version of the middleware for your operating system as you can download it from https://eid.belgium.be/