Fedict/eid-mw #42: eid-viewer: error while loading shared libraries: libeidviewer.so.0: cannot open shared object file: No such file or directory

stroobandt, 2017-08-03:
I closely followed and completed the build and install instructions on https://github.com/Fedict/eid-mw/.
My operating system is 64bit Ubuntu LTS 16.04.

Running `eid-viewer` from the command line results in the following error:

    eid-viewer: error while loading shared libraries: libeidviewer.so.0: cannot open shared object file: No such file or directory

Nonetheless, `libeidviewer.so.0` is available as a link in `/usr/local/lib`.

However, during `sudo make install`, I noticed the following warning:

    …
    libtool: warning: relinking 'libeidviewer.la'
    libtool: install: (cd /home/serge/Downloads/bin/eid-mw/plugins_tools/eid-viewer; /bin/bash "/home/serge/Downloads/bin/eid-mw/libtool"  --silent --tag CXX --mode=relink g++ -pthread -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/freetype2 -I. -I./include -Igtk -DDATAROOTDIR=\"/usr/local/share\" -I../../doc/sdk/include/rsaref220 -I../../plugins_tools/util -pthread -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/libxml2 -fvisibility=hidden -std=c++98 -g -O2 -lpcsclite -pthread -I/usr/include/PCSC -o libeidviewer.la -rpath /usr/local/lib oslayer.lo certhelpers.lo backend.lo cache/cache.lo conversions/bbannumconv.lo conversions/genderconv.lo conversions/hexnumconv.lo conversions/hexdecode.lo conversions/convert.lo conversions/convertor.lo conversions/convworker.lo conversions/dateconv.lo conversions/dob.lo conversions/doctypeconv.lo conversions/natnumconv.lo conversions/specconv.lo conversions/specorgconv.lo conversions/valdate.lo conversions/xmldoctypeconv.lo conversions/xmlspecconv.lo conversions/xmlspecorgconv.lo state.lo state-posix.lo p11.lo preview.lo xml.lo xmlmap.lo xsdloc.lo b64/base64dec.lo verify.lo ../../plugins_tools/util/liblabels.la ../../cardcomm/pkcs11/src/libbeidpkcs11.la -lssl -lcrypto -lxml2 )
    libtool: install: /usr/bin/install -c .libs/libeidviewer.so.0.0.0T /usr/local/lib/libeidviewer.so.0.0.0
    libtool: install: (cd /usr/local/lib && { ln -s -f libeidviewer.so.0.0.0 libeidviewer.so.0 || { rm -f libeidviewer.so.0 && ln -s libeidviewer.so.0.0.0 libeidviewer.so.0; }; })
    libtool: install: (cd /usr/local/lib && { ln -s -f libeidviewer.so.0.0.0 libeidviewer.so || { rm -f libeidviewer.so && ln -s libeidviewer.so.0.0.0 libeidviewer.so; }; })
    libtool: install: /usr/bin/install -c .libs/libeidviewer.lai /usr/local/lib/libeidviewer.la
    libtool: install: /usr/bin/install -c .libs/libeidviewer.a /usr/local/lib/libeidviewer.a
    libtool: install: chmod 644 /usr/local/lib/libeidviewer.a
    libtool: install: ranlib /usr/local/lib/libeidviewer.a
    libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/sbin" ldconfig -n /usr/local/lib
    ----------------------------------------------------------------------
    Libraries have been installed in:
       /usr/local/lib

    If you ever happen to want to link against installed libraries
    in a given directory, LIBDIR, you must either use libtool, and
    specify the full pathname of the library, or use the '-LLIBDIR'
    flag during linking and do at least one of the following:
       - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
         during execution
       - add LIBDIR to the 'LD_RUN_PATH' environment variable
         during linking
       - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
       - have your system administrator add LIBDIR to '/etc/ld.so.conf'

    See any operating system documentation about shared libraries for
    more information, such as the ld(1) and ld.so(8) manual pages.
    ----------------------------------------------------------------------
    …

What is wrong?
Do I need to run an additional `libtool` command?
Rebooting did not help.

**Anyhow, why is `eid-viewer` no longer packaged in a `.deb` archive, like it used to be?**
There are many Ubuntu users out there who would like to change their PIN from the comfort of their home. Thanks.

Frederikus, 2017-08-04:
Hello,

Thank you for reporting these issues.
Unfortunatelly, Wouter, who is responsible for the linux build system is currently on holidays, but will be back on August 16th to handle these issues

yoe, 2017-08-16:
Hi Serge,

On 03-08-17 11:15, Serge Y. Stroobandt wrote:
> I closely followed and completed the build and install instructions on
> https://github.com/Fedict/eid-mw/.
> My operating system is 64bit Ubuntu LTS 16.04.
>
> Running |eid-viewer| from the command line [...]

Please verify that /usr/local/lib is listed in /etc/ld.so.conf (or in a
file under /etc/ld.so.conf.d). If not, then this is why your dynamic
linker can't find the library; you'll need to add it.

> However, during |sudo make install|, I noticed the following warning:
>
> |… libtool: warning: relinking 'libeidviewer.la' libtool: install: (cd [...]

This is not a warning (as can be inferred from the absense of the word
"warning" in the output), but instead just instructions on how to use
libtool in a manual way. You can safely ignore them (provided you update
your ld.so.conf as above).

> What is wrong?
> Do I need to run an additional |libtool| command?

No, that should not be necessary.

> Rebooting did not help.
> 
> *Anyhow, why is |eid-viewer| no longer packaged in a |.deb| archive,
> like it used to be?*

It is:

wouter@eidmw:/srv/repo/reprepro$ reprepro list xenial
xenial|main|amd64: beid-mozilla-extension 4.2.5v4.2.5-0xenial1
xenial|main|amd64: eid-archive 2017.4
xenial|main|amd64: eid-mw 4.2.5v4.2.5-0xenial1
xenial|main|amd64: eid-viewer 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libbeidpkcs11-0 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libbeidpkcs11-bin 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libbeidpkcs11-dbg 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libbeidpkcs11-dev 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libeidviewer-dbg 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libeidviewer-dev 4.2.5v4.2.5-0xenial1
xenial|main|amd64: libeidviewer0 4.2.5v4.2.5-0xenial1
xenial|main|i386: beid-mozilla-extension 4.2.5v4.2.5-0xenial1
xenial|main|i386: eid-archive 2017.4
xenial|main|i386: eid-mw 4.2.5v4.2.5-0xenial1
xenial|main|i386: eid-viewer 4.2.5v4.2.5-0xenial1
xenial|main|i386: libbeidpkcs11-0 4.2.5v4.2.5-0xenial1
xenial|main|i386: libbeidpkcs11-bin 4.2.5v4.2.5-0xenial1
xenial|main|i386: libbeidpkcs11-dbg 4.2.5v4.2.5-0xenial1
xenial|main|i386: libbeidpkcs11-dev 4.2.5v4.2.5-0xenial1
xenial|main|i386: libeidviewer-dbg 4.2.5v4.2.5-0xenial1
xenial|main|i386: libeidviewer-dev 4.2.5v4.2.5-0xenial1
xenial|main|i386: libeidviewer0 4.2.5v4.2.5-0xenial1
xenial|main|source: eid-archive 2017.4
xenial|main|source: eid-mw 4.2.5v4.2.5-0xenial1

I'm not sure why your ubuntu doesn't find it, but it should.

Did you install the "eid-archive" package from the eid.belgium.be
website? Did you follow the instructions on the installation page which
tell you to run "apt-get update" followed by "apt-get install eid-mw
eid-viewer"?

If so, what happens when you run 'apt-cache policy eid-mw'?

-- 
Wouter Verhelst

yoe, 2017-08-31:
Ping. If no feedback arrives about this issue, I'll have to close it again...

yoe, 2017-09-13:
no feedback received in a month now, and I don't see any issues from my side, so closing.

closed 2017-09-13 by yoe