Fedict/eid-mw #44: Command formatting on https://eid.belgium.be/en/installation-and-removal#7314

stroobandt, 2017-08-03:
On https://eid.belgium.be/en/installation-and-removal#7314 and language variants, due to the limited page width and poor command formatting, it is not always easy for the uninformed user to distinguish between commands and the continuation of commands.
```
cd
modutil -dbdir sql:.pki/nssdb -add "Belgium eID" -libfile 
/usr/lib/libbeidpkcs11.so
```
Better would be:
```
$ cd
$ modutil -dbdir sql:.pki/nssdb -add "Belgium eID" -libfile \
  /usr/lib/libbeidpkcs11.so
```
Furthermore, it would be nice to also mention the default library location for common 64bit Ubuntu systems, i.e.: `-libfile /usr/lib/x86_64-linux-gnu/libbeidpkcs11.so.0`

stroobandt, 2017-08-07:
Still better would be the following single command:

    $ modutil -dbdir sql:$HOME/.pki/nssdb -add "Belgium eID" -libfile \
      /usr/lib/libbeidpkcs11.so

By specifically stating `$HOME`, it becomes evident this needs to be done **for every user** on the system. Neither does it hurt saying that specifically. The `$ cd` command is no longer necessary by doing so.

Frederikus, 2017-08-08:
Added the '$' and widened the whitebox in which the commands are written (and used &nbsp; ) in order to prevent the line breaking.

I'll leave it up to Wouter to add the default install paths of pkcs11 modules for the main distro's here

stroobandt, 2017-08-12:
Part of the problem is your website having huge amounts of double margins left and right, which limits the width left over for the actual content. You might have a look at your web site's CSS and make it more responsive. Actually, most people reading these pages are probably _not_ using a smartphone.

yoe, 2017-08-16:
I'm not convinced of the usability of adding the prompt sign without a clear indication that it is meant as something the computer outputs, rather than something a user has to enter. The current situation may return in users erroneously entering the dollar-sign as part of the command, which is not the wanted behaviour. For that reason, I'll remove them again, at least so long as the website is not updated to add separate formatting for prompt signs (there are no plans in that direction currently, but if that changes we might add them back).

The instructions on where to find the correct location of the PKCS#11 module is decided at compile time, so it can be pretty much anything (e.g., a user who compiles the middleware herself and uses `--prefix=/opt/eidmw` would have to use `/opt/eidmw/lib/libbeidpkcs11.so` or something along those lines). For that reason, it is better to use the `about-eid-mw` application to find the correct location.

The current instructions point to the 'default' location for classic distributions, rather than any one modern location for any one modern distribution, and explain to use the `about-eid-mw` application to find the right location, complete with a screenshot on how to exactly do it. I don't think this can be improved upon, but suggestions are welcome.

Thanks for pointing out the line-break issues, which have been fixed; with that, and the above comments, I consider this issue resolved, but feel free to reopen if you have further comments.

closed 2017-08-16 by yoe