Fedict/eid-mw #46: 32bit package libbeidpkcs11-0:i386 is no longer available

stroobandt, 2017-08-06:
The [instructions to install the 32bit package `libbeidpkcs11-0:i386`](https://eid.belgium.be/nl/installatie-en-verwijderen#7307) are no longer relevant since this 32bit package is no longer provided by `eid-archive_2017.4_all.deb`.

stroobandt, 2017-08-07:
```
$ sudo apt install libbeidpkcs11-0:i386
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libbeidpkcs11-0:i386
```

stroobandt, 2017-08-07:
`libbeidpkcs11-0:i386` is required to [digitally sign BizTax](https://github.com/Fedict/eid-mw/issues/48) declarations from within Adobe Acrobat Reader.

However, this is an antiquated and cumbersome process. It would be preferable to rather employ your very same infrastructure to sign [TaxOnWeb](http://www.taxonweb.be/) declarations.

yoe, 2017-08-16:
this is incorrect, the package *is* available. See also #42.

Since it is useless to have two issues open on essentially the same problem, I'll close this one now; please do follow up in #42.

closed 2017-08-16 by yoe