Fedict/eid-mw #48: There are ZERO ways to sign a BizTax declaration on Linux !!!

stroobandt, 2017-08-06:
This issue is related to issue #46

Here are the official possibilities to digitally sign a [BizTax](https://financien.belgium.be/nl/E-services/biztax) tax declaration, published [here](http://ccff02.minfin.fgov.be/pdfhelper/html/browsers.html?lang=nl):

[image] tax declaration](https://i.imgur.com/X0eRAss.png)

# Firefox 21, really???
It is mid 2017 and Ubuntu 16.04 LTS ships with Firefox 54 now… Out of security concerns, running version 21 is not recommendable.

# Signing externally with Adobe Acrobat Reader?
The only remaining option would be signing externally with [Adobe Acrobat 9.5.5 for Linux](https://www.adobe.com/support/security/bulletins/apsb13-15.html). However as explained in poor Dutch in [point 16.4 of this FAQ page](http://ccff02.minfin.fgov.be/pdfhelper/html/faq.html?lang=nl#no_eid_certificate_in_list_linux), this requires [the 32-bit version of package `libbeidpkcs11-0:i386`](https://eid.belgium.be/nl/installatie-en-verwijderen#7307), which is [no longer made available in `eid-archive_2017.4_all.deb`](#46).

[image]

# Conclusions
- There are ZERO ways to sign a Belgian BizTax declaration using Linux!
- Belgium is a failing state, heralded for its surrealism.
- Rather set up your company in a developed country like Switzerland and pay less taxes.

stroobandt, 2017-08-07:
Why is the same digital signature infrastructure for the personal income tax declaration [TaxOnWeb](http://www.taxonweb.be/) not being used? That works flawlessly on GNU/Linux.

Frederikus, 2017-08-07:
This looks like a very old help page indeed.
We'll contact the responsible fod regarding this,

Frederikus, 2017-08-07:
Closing this as this is not eid-mw project bug

closed 2017-08-07 by Frederikus