Fedict/eid-mw #49: 4.2.11: eid-viewer logo is missing

kyrios123, 2017-08-11:
I just compiled eid-viewer 4.2.11 on linux 64-bit and the eid-viewer.png logo is used by the .desktop file is not installed anymore.

closed 2017-08-16 by yoe

yoe, 2017-08-16:
pushed a commit that fixes this problem on the v4.2 branch.

Please note though that even though tags exist on the v4.2 branch beyond the currently-released version 4.2.5, none of those tags were tested on Linux. Since our team is rather small, testing all tags on all platforms would require unreasonable development time; therefore, a tag is usually applicable to one platform only, rather than to all platforms.

In the case of Linux, the most recent supported tag is v4.2.5. All more recent ones are not released yet.

With this change, I'll probably make a new release for Linux soon, once I've confirmed that the changes work as expected.

kyrios123, 2017-08-16:
Thanks @yoe !
Good to know, will be less packaging work for me as well ! 😃 
It would be nice to add a comment when a release is done or tagged to say which platform is concerned (or at least to do it when a the release has been tested against a non-windows platform).

kyrios123, 2017-12-01:
Hi!
Just a short message to thank you, @yoe for adding the OS when you tag a release.
This is much easier for the packagers, really appreciated !

commit d42c13df (2017-08-16): Fix Makefile.am

Automake wants directories to be specied by $() rather than by @@. Fix
usage so things work properly again.

Fixes #49.
commit fa93ce1b (2017-08-16): Fix Makefile.am

Automake wants directories to be specied by $() rather than by @@. Fix
usage so things work properly again.

Fixes #49.