Fedict/eid-mw #53: Unneeded linking, much warning during the compilation

wget, 2017-10-15:
Hi Wouter,

I have received a request by users of eid-mw on ArchLinux and these are complaining there are a bunch of library linked against eid-mw executables, and these libraries are not used at all by these executables.

```
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libgdk-3.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libglib-2.0.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libpthread.so.0' by file ('usr/bin/beid-badpin')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libgdk-3.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libglib-2.0.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libpthread.so.0' by file ('usr/bin/beid-spr-changepin')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libpthread.so.0' by file ('usr/bin/beid-askpin')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libgdk-3.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libglib-2.0.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libpthread.so.0' by file ('usr/bin/beid-askaccess')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libgdk-3.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libglib-2.0.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libpthread.so.0' by file ('usr/bin/beid-spr-askpin')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libgdk-3.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libpthread.so.0' by file ('usr/bin/beid-changepin')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/eid-viewer')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/eid-viewer')
eid-mw W: Unused shared library '/usr/lib/libssl.so.1.1' by file ('usr/bin/eid-viewer')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/eid-viewer')
eid-mw W: Unused shared library '/usr/lib/libgdk-3.so.0' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libpangocairo-1.0.so.0' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libpango-1.0.so.0' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libatk-1.0.so.0' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libcairo-gobject.so.2' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libcairo.so.2' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libgdk_pixbuf-2.0.so.0' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libgio-2.0.so.0' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/bin/about-eid-mw')
eid-mw W: Unused shared library '/usr/lib/libm.so.6' by file ('usr/lib/libbeidpkcs11.so.0.0.0')
eid-mw W: Unused shared library '/usr/lib/libpcsclite.so.1' by file ('usr/lib/libeidviewer.so.0.0.0')
eid-mw W: Unused shared library '/usr/lib/libssl.so.1.1' by file ('usr/lib/libeidviewer.so.0.0.0')
eid-mw W: Unused shared library '/usr/lib/libm.so.6' by file ('usr/lib/libeidviewer.so.0.0.0')
```

Also, during compilation, we have much warnings especially reported by libtool. Here are a few example:
```
libtool: warning: remember to run 'libtool --finish /usr/lib'
[...]
libtool: warning: relinking 'libeidviewer.la'
[...]
libtool: warning: remember to run 'libtool --finish /usr/lib'
[...]
libtool: warning: 'libeidviewer.la' has not been installed in '/usr/lib'
```

As we are at the end of 2017, do you plan to use another more recent build system like cmake?

Also, when compiling with  ̀-Wunused-result` and `-Wdeprecated-declarations`, we have quite easy to fix (at first sight) issues.

Thanks for your answers,

Regards,

yoe, 2017-10-16:
There is nothing wrong with autotools? Old does not mean broken.

The libtool warnings are harmless and can be ignored. Libtool wants some commands to be run, and it sees you haven't yet run them. Airplane will run those commands when you run 'make install' later on, and all will be well.

The 'unused shared libraries' thing are a cosmetic issue. I'll take a patch of you have one, but it's really not critical (it just means we directly link something when we could've done so indirectly).

The deprecated calls will remain, because we also need to support older distributions, and the non-deprecated versions of those APIs do not yet exist there. Every time we drop one of those older distributions, however, I check which ones are still needed and drop them where sensible.

yoe, 2017-10-16:
Sigh. Stupid autocorrect...

'Airplane' should obviously have been 'automake' :)

wget, 2017-10-16:
@yoe Thanks for your answer. Ok I understand the situation. The only fixes I can do is to provide a patch for the unused variables reported by `-Wunused-result` then.

yoe, 2017-12-07:
If the unused-result patch ever happens, please open a pull request.

I believe most of those are in the test suite, at places where we drop things just before we exit the test application or some such, or in cases where there isn't much we can do about failure to remove anymore anyway. As such, those should not have any negative outcome.

Meanwhile, closing this issue.

closed 2017-12-07 by yoe