Fedict/eid-mw #55: Google Chrome on macOS 10.13 (High Sierra): no certificates shown

Frederikus, 2017-10-30:
At the moment it is not possible to authenticate with your Belgian eID on macOS 10.13 using Google Chrome. This issue is know by the Google Chrome team and they are looking into it to fix it.

You can follow this issue here:
https://eid.belgium.be/nl/aanmelden#7496

koenvanloock, 2018-01-12:
I experienced this bug too, but was able to log in after a refresh. The second time I am prompted to fill in my pin code and everything works as intended.

Frederikus, 2018-08-10:
Fixed, was due to padding bytes in cert file that confuse google chrome.
Fix will be in release 4.4.6 and above

yoe, 2018-09-12:
It turns out that was not the actual issue, so reopening this bug.

lionelbovy, 2020-01-07:
Hello,
I've approximately the same issue except that I get the following error message: ERR_BAD_SSL_CLIENT_AUTH_CERT (via http://test.eid.belgium.be/)
However, the certificates are valid when I check in the "Keychain Access" software. Moreover, it works when I use them via other browsers such as Safari & Firefox.

Besides, I've already contacted the help desk and followed the procedure they gave me but it still doesn't work. The answer I get most of the time is to install and use Firefox. Which works perfectly fine but I think it's unfortunate that one of the most used browsers, if not the most used, doesn't work.

- OS: macOS Catalina Version 10.15.2 (19C57)
- Browser: Chrome Version 79.0.3945.88 (Build officiel) (64 bits)
- eID Middleware: BEIDToken Version 4.4.14 (1)

[image]
[image]
[image]

Frederikus, 2022-09-29:
It works with the latest eID cards.

But for older cards the following issue needs to be resolved: https://bugs.chromium.org/p/chromium/issues/detail?id=777805#c4

closed 2022-09-29 by Frederikus