Fedict/eid-mw #57: Running a service to collect cardreader info throws device error

DhondtStijn, 2017-11-13:
Hello,

I just setup 2 brand new PC's same OS, same make and model,

Both windows 10

Both equipped with visual studio 2017 and the latest version of the middle ware and the latest c# wrapper from the master branch (pkcs#11 Version 1.1.6526.26112).

We have a windows service that get triggers from a website to read the e-id.
this works on one PC but not on the other. 
On the one that fails we get  the following exception.

Net.Sf.Pkcs11.TokenException: DEVICE_ERROR
   at Net.Sf.Pkcs11.Wrapper.Pkcs11Module.checkCKR(CKR retVal)
   at Net.Sf.Pkcs11.Wrapper.Pkcs11Module.FindObjectsInit(UInt32 hSession, CK_ATTRIBUTE[] pTemplate)
   at Net.Sf.Pkcs11.Session.FindObjectsInit(P11Attribute[] attrs)

If we change it to a command line app both pc's are able to read the e-ids.
Any assistance would be nice, if any additional information would be needed we like to hear it.

Thanks in advance.

Information:

The service in question is .net 4.5.2 c# application.
The OS as stated is windows 10 Version 1703 OS.Build 15063.674

Card readers used on both systems: a usb "vasco Digipass 905" and a internal "broadcom corp contacted smart card".

We followed the sdk for the logging but do not seem to get a log file, neither in service or application mode.

Kind regards

Frederikus, 2017-11-14:
Hello Stijn,

Regarding the error:
Does it happen with both card readers, or just the external one?
When the error happens, is the card reader still operational (power LED on)?
I'm asking because the most common case our pkcs11 module returns a
DEVICE_ERROR is when we get errors from the pcsc layer below stating there
is an issue communicating with the card reader.
(And we know of issues with card readers going to sleep and not awaking
anymore untill reset)

Regarding the logging:
Did you try this manual to activate pkcs11 logging:
https://eid.belgium.be/en/specific-problems#7483 ?

Wkr,
 Frederik

DhondtStijn, 2017-11-14:
Both of them have this issue.
For the external i can confirm it wakes up as both LED's  (green en red) become active.
 
We just made a windows service above the example program and it has the same problem.
The problem always occurs on   =>session.FindObjectsInit(new P11Attribute[] { classAttribute, labelAttribute });

For the logging i followed the instruction in the sdk readme. I will follow these now and upload the log  for both card readers.

DhondtStijn, 2017-11-14:
Hello,
The log files.
 For the vasco we have logging just after we plugged it in (vasco.zip) and after it went to sleep (VascoWakeup).

[VacoWakeup.zip](https://github.com/Fedict/eid-mw/files/1470098/VacoWakeup.zip)
[Vasco.zip](https://github.com/Fedict/eid-mw/files/1470099/Vasco.zip)
[broadcomm.zip](https://github.com/Fedict/eid-mw/files/1470100/broadcomm.zip)

Thanks in advance

Frederikus, 2017-11-14:
Thanks for the logs,

I'll have a look at them.
FindObjectsInit is the function where we start reading the requested data
from the card

Frederikus, 2017-11-14:
What I see is that two card readers are detected:
VASCO DP905v2.0 0
Windows Hello for Business 11

Then, when we try to select a file on the eID Card (from within a
transaction), we get following system error:

*ERROR_BAD_COMMAND*22 (0x16)

The device does not recognize the command.
Looks like the card has entered some sort of error state. (The command we
are sending is really common, no issue there)
Do you have other applications active that are talking to the eID card?

Can you try to reproduce the issue without Windows Hello for Business
active?

Wkr,
 Frederik

DhondtStijn, 2017-11-14:
We will try to disable the hello for business and report

Kind regards
Stijn

DhondtStijn, 2017-11-14:
Hi Frederik,

We tried to disable this "windows hello  thin" and according to the os it is disabled. but we still see it in the log.  

[vasco.zip](https://github.com/Fedict/eid-mw/files/1470512/vasco.zip)

DhondtStijn, 2017-11-14:
It seems this used to be a problem in the viewer.
https://groups.google.com/forum/#!topic/eid-middleware-dev/EOWeQWFDjVs

the currentviewer works so it suppresses this in some way.  As you can see in the log when the e-id viewer reads the information it doesn't mention the "hello for business"
logging from the eid software
[vasco-eidviewer.zip](https://github.com/Fedict/eid-mw/files/1470579/vasco-eidviewer.zip)

 While going  down the logging, after the e-id viewer, you will see that  the  Microsoft hello for business triggers logging. At this pint there our software is not active and neither is the e-id viewer.

is there a way we can detect or suppress this? 
This also seems to be a new feature  since Windows 10, version 1703 or later .

Frederikus, 2017-11-14:
Hello Stijn,

Indeed, the viewer has issues with it as well. (As most lickely its the eID
card or pcsc that gets messed up, so I presume everything accessing the
card will have problems).
I doubt the current viewer does not have the issue, nothing has been
changed regarding this issue. (Perhaps it is because the hello for business
is disabled?)

Are you certain that at the end of the log, it is not your software that is
still active? (it has the same process id as in the previous log).
Could it be that your application is still active from before disabling the
"windows hello for business" ?

Wkr,
 Frederik

DhondtStijn, 2017-11-14:
Hello Frederik,

Forget the remarque about the end of the log. That is a mistake from my side with a timer that wasn't disposed at the end of the program.

But the difference in the log  that is most interesting is seeing what the eid-viewer software does and what  windows service does.. 
If you look at the log for the eid software there is no mentioning of the "hello for business card " in the log. 
If the service tries to access the reader it seems this "Windows Hello for Business 11" pops up and at this point i guess cause the issue. 

We also found out that "Windows Hello for Business 11" has to do with the azure AD domains . 
The problem seems to be once this feature is triggered you can't remove it. We removed the pc from this azure domain and used the local group policy to block this windows feature but  for some reason the "reader"  is still detected. As a side note t i still want my pc to be able to join the azure ad domain. So while this is with great certainty  the reason of our current issue in the long run more and more people /companies will run against this issue.

Looking at the log I'm wondering  why the runtime opens a second connection.  (I guess this is the runtime as it is in the beid log) 

2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     SCardConnect(VASCO DP905v2.0 0): 0x0
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     SCardBeginTransaction(0xea010000): 0x0
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:       SCardTransmit(00 A4 04 0C 0C A0 00 00 01 77 .. 43 53 2D 31 35)
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:         SCardTransmit(): SW12 = 90 00
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     SCardBeginTransaction(0xea010000): 0x0
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:       SCardTransmit(80 E4 00 00 1C)
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:         SCardTransmit(): SW12 = 90 00
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     SCardEndTransaction(0xea010000): 0x0
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     SCardEndTransaction(0xea010000): 0x0
2017-11-14 09:57:30 - 11784|14176 - cardlayer - INFO:  Connected to BE eID card in reader VASCO DP905v2.0 0
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     ReaderStatus(): card inserted
2017-11-14 09:57:30 - 11784|14176 - cardlayer - DEBUG:     SCardConnect(Windows Hello for Business 11): 0x0

 I also attached the program that was used to generate the logs. When deploying the service and configuring it as **local system** we get our current issue if you just run the example exe everything works.
 

[EidSamples.zip](https://github.com/Fedict/eid-mw/files/1471035/EidSamples.zip)

DhondtStijn, 2017-11-14:
Any idea what would trigger the second connect request.

DhondtStijn, 2017-11-21:
Hello Frederik,

We will be contacting Microsoft to see if they can shed some light on the "Windows Hello for business" issue. We will keep you informed about further developments in this area.

Thanks for the assissstance already given in this matter.

Kind regards
Stijn

mbogbp, 2018-03-16:
Hi Stijn,

Is the issue resolved by now, i'm experiencing the same issue. 

thnx

Michael

DhondtStijn, 2018-03-20:
Hi Michael,

I'm sorry to say it is not fixed. 
After contacting Microsoft for this issue it seems that the only way to find the correct department for this issue is to open a paid service request.
At this point in time it is a small problem and we are not following the service request path.

Kind regards
Stijn

mbogbp, 2018-03-30:
thnx

jeroenc89, 2019-05-20:
Hi all

Does anyone has a solution for this issue?
We are also facing issues with Windows Hello while reading the eID.

I'm testing with the latest version of EidSample.sln.
I'm sure the application is reading the correct slot.

.BEID_0.log
--------------
2019-05-20 11:47:03 - 16104|16804 - cardlayer - DEBUG:     SCardEstablishContext(): 0x0
2019-05-20 11:47:03 - 16104|16804 - cardlayer - DEBUG:     SCardListReaders(): 0x0
2019-05-20 11:47:03 - 16104|16804 - cardlayer - DEBUG:     SCardConnect(ACS CCID USB Reader 0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardBeginTransaction(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:       SCardTransmit(00 A4 04 0C 0C A0 00 00 01 77 .. 43 53 2D 31 35)
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:         SCardTransmit(): SW12 = 90 00
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardBeginTransaction(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:       SCardTransmit(80 E4 00 00 1C)
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:         SCardTransmit(): SW12 = 90 00
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardEndTransaction(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardEndTransaction(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - INFO:  Connected to BE eID card in reader ACS CCID USB Reader 0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     ReaderStatus(): card inserted
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardConnect(Windows Hello for Business 14): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardBeginTransaction(0x1): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:       SCardTransmit(00 A4 04 0C 0C A0 00 00 01 77 .. 43 53 2D 31 35)
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:         SCardTransmit(): SW12 = 6A 82
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:       SCardTransmit(00 A4 04 00 0F A0 00 00 00 30 .. 13 10 01 01 FF)
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:         SCardTransmit(): SW12 = 6A 82
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardEndTransaction(0x1): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - INFO:  Connected to unknown card in reader Windows Hello for Business 14
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     ReaderStatus(): card inserted
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardStatus(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     ReaderStatus(): card stil present
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardStatus(0x1): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     ReaderStatus(): card stil present
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardStatus(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     ReaderStatus(): card stil present
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardStatus(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardStatus(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardBeginTransaction(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:       SCardTransmit(00 A4 02 0C 02 40 31)
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:         SCardTransmit(): 0x16
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardEndTransaction(0x0): 0x0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardDisconnect(0x0): 0x0 ; mode: 0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - INFO:  Disconnected from card in reader ACS CCID USB Reader 0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - DEBUG:     SCardDisconnect(0x1): 0x0 ; mode: 0
2019-05-20 11:47:04 - 16104|16804 - cardlayer - INFO:  Disconnected from card in reader Windows Hello for Business 14

p11.log
-------------
16104 16804 20.05.2019 11:47:03 C_Initialize()             | I: enter pReserved = 0000000000000000
16104 16804 20.05.2019 11:47:03 C_Initialize()             | S: Initialize this PKCS11 Module
16104 16804 20.05.2019 11:47:03 C_Initialize()             | S: =============================
16104 16804 20.05.2019 11:47:03 C_Initialize()             | I: leave, ret = 0
16104 16804 20.05.2019 11:47:03 C_GetFunctionList()        | I: enter
16104 16804 20.05.2019 11:47:03 C_GetFunctionList()        | S: C_GetFunctionList()
16104 16804 20.05.2019 11:47:03 C_GetFunctionList()        | I: leave, CKR_OK
16104 16804 20.05.2019 11:47:03 C_GetSlotList()            | I: enter
16104 16804 20.05.2019 11:47:03 C_GetSlotList()            | I: p11_lock() acquired
16104 16804 20.05.2019 11:47:03 C_GetSlotList()            | S: C_GetSlotList()
16104 16804 20.05.2019 11:47:03 C_GetSlotList()            | I: h=0
16104 16804 20.05.2019 11:47:03 C_GetSlotList()            | I: h=0
16104 16804 20.05.2019 11:47:03 C_GetSlotList()            | I: slot[0]: ACS CCID USB Reader 0
16104 16804 20.05.2019 11:47:04 C_GetSlotList()            | I: cal_token_present
16104 16804 20.05.2019 11:47:04 C_GetSlotList()            | I: h=1
16104 16804 20.05.2019 11:47:04 C_GetSlotList()            | I: slot[1]: Windows Hello for Business 14
16104 16804 20.05.2019 11:47:04 C_GetSlotList()            | I: cal_token_present
16104 16804 20.05.2019 11:47:04 C_GetSlotList()            | I: p11_unlock()
16104 16804 20.05.2019 11:47:04 C_GetSlotList()            | I: leave, ret = 0
[... 24 lines cut ...]
[... cut ...]
16104 16804 20.05.2019 11:47:04 C_GetFunctionList()        | S: C_GetFunctionList()
16104 16804 20.05.2019 11:47:04 C_GetFunctionList()        | I: leave, CKR_OK
16104 16804 20.05.2019 11:47:04 C_Finalize()               | I: enter
16104 16804 20.05.2019 11:47:04 C_Finalize()               | I: p11_free_lock()
16104 16804 20.05.2019 11:47:04 C_Finalize()               | I: leave, ret = 0

Thanks in advance

P.s. also posted this comment on https://groups.google.com/forum/#!topic/eid-middleware-dev/r-E9RCI7MBQ

Frederikus, 2019-05-21:
see also comment on https://groups.google.com/forum/#!topic/eid-middleware-dev/r-E9RCI7MBQ

Issue appears to be fixed (not yet confirmed) when using latest head of branch 4.4

SGE66, 2019-06-25:
Hi All,

should this error be solved using belgium_eid-quickinstaller_4.4.4.3838.exe ?

Frederikus, 2019-06-25:
Hello,

No, that fix is in version 4.4.18, which is currently being tested

SGE66, 2019-06-25:
Hi, 
thx for the fast reply.

Any release date ?

On a side note our software is displaying the 4.4.4 version  as 4.3 version number.
Has something been changed in retrieving the version number of the eID Middleware?
If not, we need to check why this is  reported as 4.3 in our logfiles iso 4.4.

Stefan

Frederikus, 2019-06-25:
Hello,

Release (first as beta) will most lickely be after the taxonweb period (mid Jul)

Nothing changed regarding numbering
In e.g Windows 10 "Apps and features" it should be displayed as : "Belgium e-ID Middleware 4.4.4 (build3838)" 
and when clicking on it for more info you get the unofficial buildnumber, that does contains the revisionnumber (4.4.3838).

Wkr,
 Frederik

yoe, 2019-06-25:
I think he's asking about the C_GetInfo call, which indeed returns 4.3 in v4.4.4:

https://github.com/Fedict/eid-mw/blob/1404f856a52e32f1445b6a7e37a1664fd434f871/cardcomm/pkcs11/src/general.c#L183

This has been fixed since, and the v4.4.18 tag exports the correct version now. Additionally, the test suite on the master branch has been updated to verify that the version number in the C_GetInfo call is correct, so this should not happen again in the future.

SGE66, 2019-06-25:
> hich indeed returns 4.3 in v4.4.4:

So is this 'as-designed' or a bug (as now we do not know which middleware version is installed) ?
And will it return 4.4 in the new release ?

Stefan

yoe, 2019-06-25:
It was a bug, it should have said 4.4. Like I said, it's been fixed now and a test added to the test suite so the problem should not happen again.

Frederikus, 2019-09-16:
> Hello I am facing this DEVICE_ERROR since a while now, only one client had
> this, and now severals have.
> So, I contributed to the developpement of an exe in C# to read the id card
> to send the infos to a webApp.
> All was good until they made a windows update (exacly the win10 version
> 1803). From there, impossible to read the card ;
> any idea what windows changed to make this not working anymore?
> I have tried every versions of the eid viewer.
> here is the error;
>
> Net.Sf.Pkcs11.TokenException: DEVICE_ERROR
>    à Net.Sf.Pkcs11.Wrapper.Pkcs11Module.checkCKR(CKR retVal)
>    à Net.Sf.Pkcs11.Wrapper.Pkcs11Module.FindObjectsInit(UInt32 hSession,
> CK_ATTRIBUTE[] pTemplate)
>    à Net.Sf.Pkcs11.Session.FindObjectsInit(P11Attribute[] attrs)
>    à EidSamples.ReadData.GetData(String label) dans
> C:\inetpub\wwwroot_11_1\customs\brulabo\EfficyBeId\EfficyBeId\ReadData.cs:ligne
> 285
>    à EfficyBeId.EfficyBeId.btnReadCard_Click(Object sender, EventArgs e)
> dans
> C:\inetpub\wwwroot_11_1\customs\brulabo\EfficyBeId\EfficyBeId\EfficyBeId.cs:ligne
> 80
>    à System.Windows.Forms.Control.OnClick(EventArgs e)
>    à System.Windows.Forms.Button.OnClick(EventArgs e)
>    à System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
> [... 66 lines cut ...]
> mscorlib.resources
>     Version de l'assembly : 4.0.0.0
>     Version Win32 : 4.8.3752.0 built by: NET48REL1
>     CodeBase :
> file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/mscorlib.resources.dll

Hello,

We'll try to release a new eID Viewer for Windows in the upcomming weeks,
that has improved handling (ignoring) smart cards that are not eID cards
(Like the Hello for Business emulated cards).
So this version might fix the issues you are seeing when using Hello for
Business.

I'll post here when the software is released.

Frederikus, 2019-10-16:
We released a new eID Viewer for Windows (version 4.4.23: https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer) as beta.
This version is better at ignoring non-eID cards, like the Hello for Business software emulated cards.

SGE66, 2019-12-10:
Hi,
we encounter a similar problem with an AZURE Domain account:

Environment:
- Windows 10 Build 1909
- Belgium e-ID middleware 4.4.4 (build 3838) Laatste versie
- Belgium e-ID viewer 4.4.14 (build 4135) Laatste versie

Following error occurs when trying to read eIDCard:
ERROR: Cannot Read Carddata: Open Slot Failed. 

This problem ONLY occurs when the user logins in on his PC using a AZURE Domein account.
When a LOCAL account is used there is no issue.

Anyone any ideas whether or not this is the same problem as this issue (Windows Hello for Business blocks cardreader) ?

Regards,
Stefan

Frederikus, 2021-03-31:
Closing this. No new cases regarding Hello For business received during the last year.
Should be fixed by the improved handling for non-eID cards

closed 2021-03-31 by Frederikus