Fedict/eid-mw #59: System.Exception: 'C_GetFunctionList could not be found'

TomStevens7533, 2017-11-29:
Hi 

I'm trying to use the BEID program in a WPF but it keeps crashing at this point.
[image]

Sincerly 
Tom

yoe, 2017-11-30:
For help with developing, please contact the mailinglist at https://groups.google.com/forum/#!forum/eid-middleware-dev

This issue tracker is only for confirmed bugs in the middleware, which this is not.

closed 2017-11-30 by yoe

Rikj000, 2023-12-12:
Same question

yoe, 2023-12-12:
> Github offers a `questions` tag for these kind of issues.

... which does not mean we _have_ to use it. We preferred to create a mailinglist instead for questions related to development. There is no need to "track" questions until they're answered, and a mailinglist is a more free-flowing way of answering things like that.

I'm not saying we don't want to help developers who are having trouble understanding how the bits fit together (otherwise we wouldn't have created that mailinglist), but I don't think it's wrong of us to request that things are done through the correct forums, otherwise things get very messy.

The mailinglist still exists, on the same location as it always has.