Fedict/eid-mw #60: URGENT - YES - Always popup kill?

shamun, 2017-12-14:
When i use `Belgium eID-QuickInstaller 4.2.2.3170.exe` for my EID compiled application i do not get YES - Always popup in past.

But when i use latest  Belgium eID-QuickInstaller from official release https://eid.belgium.be/en or https://eid.belgium.be/sites/default/files/software/belgium_eid-quickinstaller_4.3.2.3551.exe , my EID compiled application always has "YES - always" popup and funny is that YES - Always is not remembered for next time which could have resolved the issue also but that Popup keep showing up on every attempt. 

Where can i get the latest Belgium eID-QuickInstaller which does not ask me the YES - Always popup please?

Frederikus, 2017-12-14:
That popup has not been brought back (see cardcomm\pkcs11\src\object.c The AllowCardReading() function is not called anymore ). Besides, if it were, it would also show up in our eID Viewer, which uses our pkcs11 library.

Could it be that your application picks up an old version of our pkcs11 library somwhere ( your application folder, somewhere in PATH, in system folder) ?

shamun, 2017-12-14:
In my application here is the code: https://github.com/shamun/eid-mw/blob/828ccf2379db638c4b2c066db9b9390d6c572fd2/doc/sdk/examples/C/src/data_more.c#L49  

I am just using default. When i use Belgium eID-QuickInstaller latest/current or beta i get that popup. Until i manually install this Belgium eID-QuickInstaller 4.2.2.3170.exe the popup keeps opening.

shamun, 2017-12-14:
Seems like in my OLD compiled version i had AllowCardReading()  enabled. And that might be causing the YES - Always popup triggered. WOW!!

https://github.com/shamun/eid-mw/blob/828ccf2379db638c4b2c066db9b9390d6c572fd2/cardcomm/pkcs11/src/object.c#L369

Frederikus, 2017-12-14:
You're Welcome,

So you're using your own build of beidpkcs11.dll (that showed the dialog), which got overridden by our beidpkcs11.dll again when you ran the 4.2.2 installer.

Out of curiosity: what is the reason for using your own version of beidpkcs11.dll instead of just running your application against the one from our installer?

closed 2017-12-14 by Frederikus