Fedict/eid-mw #62: Strange ERROR - any idea why its failing?

shamun, 2018-01-22:
Intel NUC i5 PC, i was installing Eid to read the id card. But i am getting strange error as below with official Eid Viewer and drivers, also same with future release beta version.

[image]

[image]
```
W: unknown eidviewer_log_level setting identifier found in registry: 
W: Known values are: Error, Warning, Info, Debug
W: switching viewer log level to the default setting (Error)
E: eid_vwr_convert_set_langSystem.BadImageFormatException: Poging om een programma te laden met een onjuiste indeling. (Uitzondering van HRESULT: 0x8007000B)
   bij eIDViewer.NativeMethods.eid_vwr_convert_set_lang(eid_vwr_langs lang)
   bij eIDViewer.NativeMethods.ChangeLanguage(eid_vwr_langs language)
```

When i ran my old compiled eid.exe it get following error: 
```
c:\eid.exe
beidpkcs11.dll not found
err is 0x000000c1
reader_unplugged
```

shamun, 2018-01-22:
Cant uninstall in general way!

[image]
[image]
[image]

shamun, 2018-01-22:
Intel NUC PC, do not have any USB 2.0 interface. USB 3.0 blue and  yellow. Does it mean the EiD drivers are not compatible yet for USB 3.0 blue interface and yellow interface?

[image]

Frederikus, 2018-01-23:
Hello Shamun,

We do not write, nor install USB or card reader drivers.
When communicating with the eID card, the drivers you installed on your OS will be used (and not even directly by us)

As for the eID Viewer launch error:
It appears that the 32-bit version of the beidpkcs11.dll cannot be found, and the system afterwards tries to load the 64-bit version (which then fails due to wrong architecture).
Can you confirm the correct dll's are present in the same folder as the eID VIewer.exe?

For the uninstall error:
Did you ran some cleanup tool that removed the uninstall .msi, but did not alter the windows registry?
see FAQ: https://eid.belgium.be/en/advanced-problems#7277

Wkr,
 Frederik

shamun, 2018-01-23:
Hello Frederik,

Please find my feedback below:

> We do not write, nor install USB or card reader drivers.
> When communicating with the eID card, the drivers you installed on your OS will be used (and not even directly by us)

1. Windows 10 Pro was brand new and the Intel NUC PC was brand new. Without any modification i have for the first time connected the card reader OMNIKey-3121. Windows 10 Pro automagically detected the reader and in device manager it shows the card reader is detected, has no issue found.

[image]

> As for the eID Viewer launch error:
> It appears that the 32-bit version of the beidpkcs11.dll cannot be found, and the system afterwards tries to load the 64-bit version (which then fails due to [...]

2. YES - i have done no modification, i did first step went to EID official site and installed the quickinstaller and viewer. On execute they failed. I again verified if files were missing but i could not see any abnormal file missing. Nor i am done any manual activity to make any file missing. So it was out of the blue like experience to me in that PC, when  eID viewer were failing

> For the uninstall error:
> Did you ran some cleanup tool that removed the uninstall .msi, but did not alter the windows registry?
> see FAQ: https://eid.belgium.be/en/advanced-problems#7277

3. NO - i have not done any cleanup procedure. The PC and the hardware came to me for installing eID card software on it and test it. Without any modification i had that experience. 

4. For your information the uninstallation worked after reboot.

I removed Windows 10 Pro and then re-installed the same Windows 10 Pro on that same hardware. Then i went to official eID site and re-installed eID quickinstaller and viewer. Immediately it started to work.

I wonder why and what went wrong but its clear to me some file corruption occured in first installation before re-installing the windows 10 pro.

Thanks and regards
Shamun

closed 2018-01-23 by shamun