Fedict/eid-mw #65: Reading identity information through the Firefox addon

michaelshmitty, 2018-01-25:
From the wiki:
"Note that there are also some alternative PKCS#11 implementations in existence which support the Belgian eID. While these alternative implementations support the cryptographic operations, they usually do not support the identity information. Therefore, they cannot be used to read all data off the card."

Is the PKCS#11 implementation in Firefox that the eid addon uses such an alternative implementation as described above? Does this mean that the Firefox eid addon will not be able to read anything other than cryptographic information?
I'm trying to find out if I can have a web application read identity information through the firefox addon or that it can only be used for authentication and authorization.
Thanks!

yoe, 2018-01-25:
That wiki refers to (at least) [OpenSC](/opensc/opensc), which has support for the Belgian electronic ID card. There may be other PKCS#11 implementations that we don't know of.

The official PKCS#11 module supports reading data through `C_GetAttributeValue` calls on certain objects of type `CKO_DATA` that are implemented by it, as explained in [the documentation](https://github.com/Fedict/eid-mw/blob/master/doc/sdk/documentation/beidsdk_card_data.pdf). This is very much a BeID-specific operation, however, and is not supported by Firefox (which only cares about accessing the certificates on the card). As such, even in contexts where you *can* access the PKCS#11 modules (mostly the [browser.pkcs11 API for add-ons](https://developer.mozilla.org/en-US/Add-ons/WebExtensions/API/pkcs11), reading identity data from the card is not possible.

michaelshmitty, 2018-01-25:
> As such, even in contexts where you can access the PKCS#11 modules (mostly the browser.pkcs11 API for add-ons, reading identity data from the card is not possible.

Exactly what I needed to know. Cheers!

closed 2018-01-25 by michaelshmitty