Fedict/eid-mw #66: Debian package 2017.6 seems broken or corrupt

bertvandepoel, 2018-01-30:
Over a week ago, the update for 2017.6 was pushed to the debian repository. However, this update consistently fails on any system I've checked. Also a manual download of the 2017.6 deb package yields the same problem. Specifically, I'm getting this error:

`
bert@berts-leuven-desktop:~$ sudo apt-get dist-upgrade
Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd       
De status informatie wordt gelezen... Klaar
Opwaardering wordt doorgerekend... Klaar
De volgende pakketten zullen opgewaardeerd worden:
  eid-archive
1 pakketten opgewaardeerd, 0 pakketten nieuw geïnstalleerd, 0 te verwijderen en 0 niet opgewaardeerd.
14 pakketten niet volledig geïnstalleerd of verwijderd.
Er moeten 0 B/7728 B aan archieven opgehaald worden.
Door deze operatie zal er 0 B extra schijfruimte gebruikt worden.
Wilt u doorgaan? [J/n] 
dpkg-deb: fout: in archief '/var/cache/apt/archives/eid-archive_2017.6_all.deb' komt element 'control.tar.xz' voortijdig voor 'control.tar.gz'; er wordt opgegeven
dpkg: fout bij verwerken van archief /var/cache/apt/archives/eid-archive_2017.6_all.deb (--unpack):
 subproces dpkg-deb --control gaf een foutwaarde 2 terug
Fouten gevonden tijdens behandelen van:
 /var/cache/apt/archives/eid-archive_2017.6_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`
Also trying to open the deb-file manually, it seems that the file can't be opened and is considered corrupt. 

I think someone either made a mistake while building the debian package, or the upload was somehow incomplete.

Just for reference, I run Ubuntu 14.04 trusty amd64.

Frederikus, 2018-01-31:
Hello Bert,

Thank you for reporting
That's Wouter who has been handling those, I'll let him know as soon as I
see him (he has a few days off).

For the meantime, you can always find the previous package at:
https://eid.belgium.be/sites/default/files/software/eid-archive_2017.4_all.deb

Wkr,
 Frederik

yoe, 2018-01-31:
It turns out that Debian Unstable recently enabled xz compression for the package metadata, which requires dpkg 1.17.6 or higher to install; Ubuntu 14.04 has 1.17.5, so that's too old.

I'll fix this next week.

bertvandepoel, 2018-01-31:
Aha, that makes a lot of sense. Thank you for looking into it!

yoe, 2018-01-31:
FWIW, the fix is 085e381f39885a383002df8f2e00da33ea36e2f1 in Fedict/eid-archive, and has been released onto the website, but the repository will have to wait until next week

yoe, 2018-02-07:
This is now uploaded to the public repository, so closing the bug.

closed 2018-02-07 by yoe