Fedict/eid-mw #67: Talk to middleware from webbrowser

cimm, 2018-02-12:
As a Belgian citizen I can use my Belgian eID card to authenticate myself on the [CSAM](https://iamapps.belgium.be/tma/?lang=en) website after [configuring Firefox](https://eid.belgium.be/en/faq/install-and-activate-eid-add-firefox#7507) to use the [eid-mw](https://github.com/Fedict/eid-mw) middle-ware. I didn't need any Firefox extensions for this, great!

I am guessing the CSAM website uses some JavaScript magic to talk to the eID card when using their login form. The eID middleware catches this and handles the communication with the eID card.

Problem is I don't see how to do this as a web developer. How can my own website ask the eID middle-ware to read the card in the card reader? I know how to this from software, on the machine itself, but not from within a browser.

I could write my own Firefox extension that talks directly to the middle-ware but that seems cumbersome since everyone would need to install uncountable different Firefox extensions to use their Belgian eID card and it's obviously possible without it since that's exactly what the CSAM website is doing.

Is there any example code available or documentation available that could show me how I can talk to the eID middle-ware from within a browser?

Thank you!

lsaffre, 2018-02-12:
Simon, I hope that you get better feedback than the following one.
I wrote a Java applet (JNLP application) to do this:
http://lino-framework.org/eidreader/index.html
And that applet is still being used on a few Lino sites, but it requires
some older web browser because the newer ones don't have Java anymore.
If you find some better solution, please let me know.

You say "I know how to this from software, on the machine itself" :
please tell me how you do that.

Luc

cimm, 2018-02-12:
Thank you Luc, everything I find on the interwebs is using some Java applet. There have to be better ways, no? The Java applets days are clearly over and the CSAM website is not using an applet as far as I can tell.

To answer your question in return... I used the [graphene](https://github.com/PeculiarVentures/graphene) library to talk to the eid-mw. You can find a sample script in #61.

metsma, 2018-02-12:
Estonian government solutions:
https://github.com/open-eid/browser-token-signing
https://github.com/open-eid/chrome-token-signing
https://github.com/hwcrypto/hwcrypto.js

yoe, 2018-02-13:
Hi Simon,

On 12-02-18 15:57, Simon Schoeters wrote:
> As a Belgian citizen I can use my Belgian eID card to authenticate
> myself on the CSAM <https://iamapps.belgium.be/tma/?lang=en> website
> after configuring Firefox [...]
This is incorrect.

The eID middleware is a "security module" in Firefox. All the addon does
is install it; it is firefox which then handles it, much like any
client-cert TLS session.
> Problem is I don't see how to do this as a web developer. How can my
> own website ask the eID middle-ware to read the card in the card
> reader? I know how to this from software, on the machine itself, [...]
Actually it isn't.

The CSAM website gets the certificate and parses it. The certificate DN
has the given and last names of the card holder, as well as their
national registry number.

Since the government is allowed by law (under certain conditions) to
contact the national registry and get more information, this is what
they do; but obviously you would not be able to do so.

> Is there any example code available or documentation available that
> could show me how I can talk to the eID middle-ware from within a browser?
>
These are the options I know of:

  * If you need an indication of personal information (as opposed to a
    strong proof of identity), you can use the drag-and-drop
    functionality from the eid-viewer (just drag the photo onto a
    webpage, that drops the XML representation of the card). Note though
    that the XML representation is not signed, so therefore it is
    trivial to forge; as such, this is not a reasonable option if you
    need proof of identity. For a proof of concept on how this works,
    see https://github.com/Fedict/eid-viewer/tree/master/xml
  * If you need nothing more than a person's name or national registry
    number, then have them authenticate against your webserver and parse
    the certificate. This also works for things that are easily read
    from the national registry number, such as the date of birth and the
    gender.
  * You can contact a third-party identity provider, who will have the
    necessary infrastructure available to read identity information from
    the card, and will provide you with all the information necessary.
  * Some work has recently been done on the eID applet to make it
    possible to run it as a Java Webstart application (i.e., a JNLP
    thing). I don't know how well that works right now (I don't work on
    the applet), you'll have to check for yourself
  * Finally, you can go the extension way; but indeed that's not ideal,
    for reasons that you've pointed out yourself.

cimm, 2018-02-13:
@yoe Thank you for you extensive reply. I learned a lot from it!

I am a bit disappointed it's not easier to use eID cards in our web applications but I am super impressed by how I can find open source code by our government on Github with such a responsive team behind it. Head of to the Fedict team for that!

cimm, 2018-02-13:
@metsma Thanks. I ran into the Estionian project as well. Did you manage to get the Belgian eID working with the hwcrypto.js library? It doesn't seem to recognize the card for me (not sure yet if it's the card or my card reader).

metsma, 2018-02-14:
It should work out of the box with windows drivers.
OSX/Linux needs PKCS11 driver paths and card ATR-s
https://github.com/open-eid/chrome-token-signing/blob/master/host-shared/PKCS11Path.cpp#L103
Not sure if it Belgium cards work with OpenSC middleware.

yoe, 2018-02-14:
@metsma they do, but there are a few limitations:

- OpenSC has no support for the signature certificate, only for the authentication certificate;
- This eID middleware has support for reading identity information from the card using `CKO_DATA` read calls through the PKCS#11 API. The OpenSC PKCS#11 module does not implement this.

Since this is not a bug in the middleware, closing this issue now. @cimm, please note that there is a mailinglist for questions about the middleware, at https://groups.google.com/forum/#!forum/eid-middleware-dev (for development) and https://groups.google.com/forum/#!forum/eid-mw (for general middleware-related questions).

closed 2018-02-14 by yoe

yoe, 2018-02-15:
@lsaffre I absolutely agree that it's a workaround and not a good solution, and that it won't work everywhere. It's the best we could do on a limited budget, however.

julesdehenin, 2018-11-15:
@yoe you said

> If you need nothing more than a person's name or national registry
>     number, then have them authenticate against your webserver and parse
>     the certificate. This also works for things that are easily read
>     from the national registry number, such as the date of birth and the
>     gender.
This sounds good enough for my web application but how do I get my php script to access the certficates known to Firefox. Any example code would be great.

yoe, 2018-11-15:
@julesdehenin that depends on the webserver in use, since it is the webserver (and not php) that does the parsing. Apache, for instance, exposes them in server variables.

Try using [phpinfo()](https://secure.php.net/manual/en/function.phpinfo.php) from a PHP page that is set up to require Certificate authentication from the Belgian eID roots. That should give you all the information you need.

julesdehenin, 2018-11-15:
Do I understand it correctly if I say that the user has to authenticate against the webserver running the php script and the way to authenticate is via the certificate on the eID?
In case my conceptual description is correct, I'm still having trouble writing a php script that would do that. I would think this would be a combination of setting some Apache settings and writing a few lines of php to prompt for the certificate and then read and parse the content of the certificate.

yoe, 2022-03-28:
> 4 years have passed since the original post, did anyone find a suitable solution for this? My use case is similar to @lsaffre 's: pharmacists need to read card info from their patient's eID card (more than just name/personal id number). This needs to happen fast and without the need for patients to authenticate.
> 
> Drag & drop from eID viewer works, but as others pointed out, it is cumbersome for the user. Is there a better solution available?

There is currently no generally-available "better" solution, but there are two options that should be easier to use than the drag-and-drop application

You can either use one of the commercial identity providers, or you can write an add-on which uses a small native application to read data from the card and inject it into your webpage.

cliffdeclerck, 2022-03-28:
@yoe Hi Wouter, thanks for the reply! Could you briefly indicate which commericial solutions there are on the market?

Utopiah, 2022-08-12:
@fikkatra I didn't dive in depth but you might want to try https://demo.eideasy.com/login-widget and then if this workflow fits your need check the documentation and pricing. Closest to a "JavaScript solution" I found so far.

I wish there was a NodeJS backend with a JS frontend with a Dockerfile to do so directly but didn't stumble upon that yet.

fikkatra, 2022-08-12:
> @fikkatra I didn't dive in depth but you might want to try https://demo.eideasy.com/login-widget and then if this workflow fits your need check the documentation and pricing. Closest to a "JavaScript solution" I found so far.
> 
> I wish there was a NodeJS backend with a JS frontend with a Dockerfile to do so directly but didn't stumble upon that yet.
> 
> Edit: found via https://groups.google.com/g/eid-middleware-dev/c/_ov6KeHrcqo

Thanks @Utopiah , that is very helpful. As I understand, they support the _login_ and _signing_ flow (the latter through a browser plugin). Do you happen to know if they support the _reading_ flow, i.e. read data from the card without authenticating?

Utopiah, 2022-08-12:
I believe if _login_ can be done _reading_ can be done but unfortunately I didn't try so I'm not sure. Also I didn't try the _signing_ flow yet either but I hope and believe that the eID Belgium available at https://eid.belgium.be would be sufficient.

kristofvanwoensel, 2022-10-14:
Following yoe's second solution, you can use [PyPI eidreader](https://pypi.org/project/eidreader/) to post the card's data to a route on your webpage.

- you'll need to install python and the eid middleware on the client computer.
- your website needs a trigger button to execute the local python script (see documentation).
- the python script posts the data to the defined route.