Fedict/eid-mw #68: Read data from card via command-line?

lsaffre, 2018-03-29:
How can I launch a program that reads the data on the eid card without user interaction and writes it to an xml file or some other format? A straightforward approach would be some via command-line option for eid-reader, but I cannot find any documentation about such a feature.

yoe, 2018-03-30:
eid-viewer indeed does not have an option to write the XML file without user interaction. We did consider that, but saw it as feature creep, and therefore decided against it.

However, the XML generating code is part of the eID Viewer library, so writing a short program to do so (or integrating that library in and existing program) should be fairly straightforward.

closed 2018-03-30 by yoe

lsaffre, 2018-04-16:
I wrote and published a short program which does so: http://eidreader.lino-framework.org/
(Actually this one uses JSON format instead of XML and has some features for integrating it into a web application using a custom url schema handler)