Fedict/eid-mw #70: Fonctionnait en janvier, ne fonctionne plus en avril

tierce, 2018-04-16:
Bonjour,

Le même ordinateur, le même lecteur de carte intégré (Gemalto) et la même carte d'identité ont été utilisé pour me connecter et signer ma déclaration TVA le 18 janvier 2018.

Cet ordinateur équipé de cette saloperie de Microsoft® Windows® 10 sert principalement à la signature de ma déclaration TVA puisqu'il m'est impossible de le faire avec mon GNU/Linux. Autant dire qu'il me sert tout les trois mois quand bien même il soit allumé dans un coin pour faire ses mise à jours.  Ce qu'il fit encore ce 11 avril 2018.

J'ai une nouvelle fois pris le temps de :

- désinstallé le middleware
- désinstallé le viewer
- désinstallé le lecteur de carte Gemalto
- redémarré (shutdown /g /t00) pour être sur que ce soit un « vrais » redémarrage
- laisser Windows® détecter et installer de lecteur de carte
- désactivé le lecteur de carte Gemalto pour simuler un « débranchement »
- installé le middleware belgium_eid-quickinstaller_4.3.2.3551
- activé le lecteur au moment ou l'assistant demande de le « brancher »
- il ne le détecte pas alors qu'il est bel et bien dans le gestionnaire de périphérique, sans erreur et activé
- cliqué sur terminé
- installé le viewer belgium_eid_viewer_installer_4.2.11.3344

le résultat étant [la capture suivante](https://computhings.be/documents/index.php/apps/gallery/preview.public/3745?width=1300&height=1300&c=f705b66e16b616b763e9dae5bebd9cc2&requesttoken=hh%2BtSLFHfxWL1ddK9r7UYGPxHaf7SU1eOxoHHLHlx3E%3D%3A0kfgP%2BkOCFzzluMhvZHhJiLHb8CVG2I4QWAwU%2Fe%2F9Tw%3D&token=SWuE1lHLo6XkXpM)

J'ai refait la même chose avec un ACR38U, à la différence près qu'il n'est ni reconnu durant l'installation du middleware, ni reconnu, même sans nom comme le lecteur intégré ([voir capture](https://computhings.be/documents/index.php/apps/gallery/preview.public/3745?width=1300&height=1300&c=f705b66e16b616b763e9dae5bebd9cc2&requesttoken=hh%2BtSLFHfxWL1ddK9r7UYGPxHaf7SU1eOxoHHLHlx3E%3D%3A0kfgP%2BkOCFzzluMhvZHhJiLHb8CVG2I4QWAwU%2Fe%2F9Tw%3D&token=SWuE1lHLo6XkXpM)).
Cela fonctionne donc encore moins avec le ACR38U.

Le eID Viewer, le ACR38U et ma carte d'identité fonctionnent sur mon GNU/Linux Trisquel.
Mais Adobe® ne proposant aucune version libre pour apposer ma signature, je suis dépendant de ces autres outils privateurs qui, malgré tout mes efforts ne me permettent même pas de me connecter, ni donc de signer ma déclaration TVA.

Merci de bien vouloir m'accompagner pour trouver une solution.

Cordialement,

Frederikus, 2018-04-17:
Hello,

About card reader ACR38U, the non-CCID version:

There are known issues with a new driver version (4.2.6.0) of this reader.
(i.e. no connection to the card reader is possible with this driver version)
Versions 4.2.0.0 and 4.0.6.0 have been tested by us and did not gave the
issue above.
So if you could try to downgrade your ACS driver to one of those older
versions?

The configuration menus for device drivers can be found here:
https://support.microsoft.com/fr-be/help/4028443/windows-update-drivers-in-windows-10

For more in depth support, please contact our service desk at:
https://eid.belgium.be/fr/contact

Wkr,
 Frederik

PS:
In the steps you described, it is not needed to do all these deinstalls
Our new installer will uninstall its previous versions, and you can keep on
using the working Gemalto driver. (or update it, but that's independant of
the eID software)

It is also not needed to unplug and re-plug the card reader (and for sure
not to disable its driver) during the installation test, all card readers
that are present when you press the next button should be found.

tierce, 2018-04-17:
I tried with the 4.0.6.0 from [ACS website](https://www.acs.com.hk/download-driver-unified/7289/ACS-Unified-MSI-4060.zip) (pilote 4.0.0.6) and 4.0.7.0 from [BeID website](http://www.belgeid.be/fr/product/acr38) without success.

Reader detected but no card readed.

More strange, REMOVING both my card readers (Gemalto and ACR), the eID Viewer still consider that there is a card reader (no blue exlamation mark on the bottom left corner of the application), one « empty » entry in the Files > Card Readers > [    [      ]          ] as you can see on the capture named LecteurDeCarteFantome.PNG on [this capture collection](https://computhings.be/documents/index.php/s/b5adtbEqeg4dSl7).

Am I doomed or something? :smiling_imp:

tierce, 2018-04-18:
The problem is not solved ... it is a work around.

**More detailed situation...** 

I use a Microsoft® Windows® Acer Travelmate laptop with a dead monitor.
In January 2018, I was using a VGA display to resurrect this unused resource in the idea to keep it under the hand (remotely) ONLY for the PDF signatures of may VAT declaration.

So yes, in January 2018 the Gemalto internal ID reader was working without problem.

All the [current issues I had](https://computhings.be/documents/index.php/s/b5adtbEqeg4dSl7) are related to the fact that I use [Remmina](https://packages.debian.org/jessie/remmina) as the [Remote Desktop Protocol](https://en.wikipedia.org/wiki/Remote_Desktop_Protocol) client from my daily usage computer running [Trisquel GNU/Linux](https://trisquel.info/).

I do not know if the situation would be the same with another RDP client.

So, **to have it working, I have to plug a display and do it NOT remotely**.

I'm not doomed :smiling_imp: , I'm just a guy trying to keep it's computers hardware and softwares as close to Free and Open Source as possible :expressionless: .

yoe, 2018-04-19:
The RDP protocol has the ability to forward PC/SC messages through the connection, allowing you to use the smartcard you've connected to your local system. It may be possible that a Windows update now prevents you from using the smartcard connected to the local machine through an RDP connection.

There are some RDP clients for GNU/Linux which support that mode; e.g., "freerdp-x11" has a "/smartcard" option which allows you to do that.

At any rate, what you're trying to do is not a bug in our middleware; even if it was, this is not a supported mode of operation. Therefore, I'm closing this issue.

closed 2018-04-19 by yoe