Fedict/eid-mw #71: MacOS App - MW Integration Questions.

r9zv9n, 2018-04-25:
Hello, 

I’m working on a MacOS app that: 

A) Reads information off of the card (e.g., Signature certificate, Citizen CA certificate, Root CA certificate, Identity, Address, Photo, Identity Signature, Address Signature, NationalRegistry Certificate), and then 

B) Signs some digest data. 

Please correct me if I'm wrong: 

For A), one can adopt eIDOSLayerUI and use eIDOSLayerBackend to retrieve the above data (as per the eID Viewer project). Can one use eIDOSLayerUI/eIDOSLayerBackend to retrieve *all* of the above data (including Identity & Address Signatures)? Will these be labelled “sign_data_file” and “sign_address_file” respectively, as per https://github.com/Fedict/eid-mw/blob/master/doc/sdk/documentation/beidsdk_card_data.pdf? 

For B), there is no high-level ObjC wrapper APIs to sign data, is this correct? If so, does one have to use the C APIs as per the "eid-mw/doc/sdk/examples/C/src/sign.c” sample? If so, will opening a session for B) interfere with the session that's opened for A)?  

Thank you.

yoe, 2018-04-25:
Hi,

On 25-04-18 10:51, r9zv9n wrote:
> Hello,
>
> I’m working on a MacOS app that:
>
> A) Reads information off of the card (e.g., Signature certificate,
> Citizen CA certificate, Root CA certificate, Identity, Address, Photo,
> Identity [...]
Yes.
>
> Will these be labelled “sign_data_file” and “sign_address_file”
> respectively, as per
> https://github.com/Fedict/eid-mw/blob/master/doc/sdk/documentation/beidsdk_card_data.pdf?
>
Yes, the names have not been changed from the PKCS#11 names as per that
document.
>
> For B), there is no high-level ObjC wrapper APIs to sign data, is this
> correct? If so, does one have to use the C APIs as per the
> "eid-mw/doc/sdk/examples/C/src/sign.c” sample? If so, will opening a
> session for B) interfere with the session that's opened for A)?
>
Indeed, there isn't a high-level ObjC wrapper. The eID viewer API was
written mostly for the eID viewer, which doesn't do any signatures.
Additionally, signing data is somewhat fiddly, which may make it harder
to use a higher level API like that.

The PKCS#11 API allows you to create multiple sessions however, so that
should be safe. As soon as all the data is read (which you can find out
by way of the state machine and the "new source" events), the backend
should not access the card anymore (unless you tell it to), so then it
should be safe to create a new session and to directly call C_SignInit()
etc.

r9zv9n, 2018-04-25:
Great, will do that, thank you.

yoe, 2018-04-25:
Okay.

Since this isn't actually a bug report, closing this now.

Please note for future reference that we have a developers' forum at https://groups.google.com/forum/#!forum/eid-middleware-dev

closed 2018-04-25 by yoe