Fedict/eid-mw #72: eid-viewer heeft als vereiste libcurl3 (>= 7.16.2)

JenKD, 2018-06-01:
Hi, 
eid-viewer in Debian testing heeft als vereiste libcurl3 (>= 7.16.2).  In debian testing werd libcurl4 geïntroduceerd:  hierdoor kan ik eid-viewer niet installeren. (versie 4.4.2v4.4.2 ).

libcurl4 is in strijd met libcurl3
cmake heeft als vereiste libcurl4 (>= 7.16.2)
clamav heeft als vereiste libcurl4 (>= 7.16.2)
apache2-bin heeft als vereiste libcurl4 (>= 7.16.2)
curl heeft als vereiste libcurl4 (= 7.60.0-2)

Johan

yoe, 2018-06-05:
Hoi Johan,

Debian testing wordt niet ondersteund, omdat dat een development-versie van de distributie is. U hebt twee mogelijkheden:

- compileer het pakket zelf:

        apt-get -b source eid-mw

    Hiervoor moet u wel eerst een `deb-src` lijn aan het bestand `/etc/sources.list.d` toevoegen (kopieer de bestaande `deb` lijnen en verander ze naar een `deb-src` variant)
- gebruik de bestaande sid (unstable) repository:

        deb https://files.eid.belgium.be/debian continuous/sid main

    Let wel: deze repository bevat bètaversies van de eid-software en kan daarom van tijd tot tijd uitzonderlijk even niet functioneren.

closed 2018-06-05 by yoe

JenKD, 2018-06-05:
Thanks Wouter,
kan je ook een Release.gpg toevoegen aan de sid-repo?

W: GPG error: http://files2.eid.belgium.be/debian continuous/sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY xxxxx
E: The repository 'http://files2.eid.belgium.be/debian continuous/sid InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://files.eid.belgium.be/debian continuous/sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY xxxxxxxxxxx
E: The repository 'http://files.eid.belgium.be/debian continuous/sid InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

TIA
Johan

JenKD, 2018-06-05:
Inderdaad, opgelost!  Hartelijk dank.

Johan

pdebuyl, 2019-04-23:
Hi @yoe thank you for the tip of rebuilding from sources. I am using a mixed stretch/sid system and some packages required libcurl4.