Fedict/eid-mw #73: EID Viewer not working on 2012R2 RDS

JoAerts, 2018-06-13:
Hi,

The current EID Viewer doesn't seem to work in an RDS environment. The old Java-version works without a problem. What logfiles can I provide to useful debug information?

Grts

yoe, 2018-06-13:
You could start by explaining the failure mode. How exactly does it fail?

Also, what is RDS? ;-)

JoAerts, 2018-06-13:
RDS is Terminal Server :-) 
Middleware works fine, login test on eid.belgium.be works, but EID-viewer doesn't show the card information. Following is the debug log:

> D: CSCblogtest called Handling state transition for event SET_CALLBACKS
> D: CSCblogtest called Leaving state LIBOPEN
> D: CSCblogtest called Entering state CALLBACKS (target)
> D: CSCblogtest called Entering state NO_TOKEN (child)
> D: CSCbnewstate called STATE_NO_TOKEN
> D: CSCbNewSrc called EID_VWR_SRC_NONE
> D: CSCblogtest called Entering state NO_READER (child)
> D: CSCbnewstate called STATE_NO_READER
> D: CSCblogtest called State transition for SET_CALLBACKS complete
> D: CSCblogtest called Handling state transition for event READER_FOUND
> D: CSCblogtest called Leaving state NO_READER
> D: CSCblogtest called Entering state READY (target)
> D: CSCbnewstate called STATE_READY
> D: CSCblogtest called State transition for READER_FOUND complete
> D: log_level combo selected
> N: CbReadersChanged called without a reader description
> D: CSCblogtest called Handling state transition for event TOKEN_INSERTED
> D: CSCblogtest called Leaving state READY
> D: CSCblogtest called Leaving state NO_TOKEN
> D: CSCblogtest called Entering state TOKEN (target)
> D: CSCbnewstate called STATE_TOKEN
> D: CSCblogtest called C_OpenSession(slot, 0x00000004, 0, 0, &session) returned 0xe1
> D: CSCblogtest called Entering state TOKEN_ID (child)
> D: CSCbnewstate called STATE_TOKEN_ID
> D: CSCblogtest called C_FindObjectsInit(session, &attr, 1) returned 0xb3
> D: CSCblogtest called Handling state transition for event STATE_ERROR
> D: CSCblogtest called Leaving state TOKEN_ID
> D: CSCblogtest called C_FindObjectsFinal(session) returned 0xb3
> D: CSCblogtest called Entering state TOKEN_ERROR (target)
> D: CSCbnewstate called STATE_TOKEN_ERROR
> D: CSCblogtest called State transition for STATE_ERROR complete
> D: CSCblogtest called State transition detected, aborting handling of TOKEN_INSERTED

yoe, 2018-06-13:
>     D: CSCblogtest called Entering state TOKEN (target)
>     D: CSCbnewstate called STATE_TOKEN
>     D: CSCblogtest called C_OpenSession(slot, 0x00000004, 0, 0,
>     &session) returned 0xe1
>

This is your problem. 0xE1 is CKR_TOKEN_NOT_RECOGNIZED; it means that
the viewer finds a cardreader with a card in, but that the card in
question is not an eID card.

Are you perchance using Microsoft Hello for Business somewhere? That
does result in this kind of problem.

Have you tried to manually select a card reader in the File menu?

JoAerts, 2018-06-13:
I'm afraid that's not even possible... no selection list.
[image]

With OpenSC, I've found out there are two card-readers

> 0    Yes   PIN pad   ACS APG8201 0
> 1    Yes             ACS CCID USB Reader 0

First one is my Isabel card reader, when I disconnect that one, EID viewer works on my Windows clients :-)

But... we also use a lot of thinclients based on freerdp. OpenSC detects the card reader and recognizes the card as Belpic card, but eidviewer does nothing (login test on eid.belgium.be works also):

> D: CSCblogtest called Handling state transition for event SET_CALLBACKS
> D: CSCblogtest called Leaving state LIBOPEN
> D: CSCblogtest called Entering state CALLBACKS (target)
> D: CSCblogtest called Entering state NO_TOKEN (child)
> D: CSCbnewstate called STATE_NO_TOKEN
> D: CSCbNewSrc called EID_VWR_SRC_NONE
> D: CSCblogtest called Entering state NO_READER (child)
> D: CSCbnewstate called STATE_NO_READER
> D: CSCblogtest called State transition for SET_CALLBACKS complete
> D: CSCblogtest called Handling state transition for event READER_FOUND
> D: CSCblogtest called Leaving state NO_READER
> D: CSCblogtest called Entering state READY (target)
> D: CSCbnewstate called STATE_READY
> D: CSCblogtest called State transition for READER_FOUND complete
> D: log_level combo selected
> W: unknown language identifier found in registry: 
> W: switching language to the system setting 
> E: CSCblogtest called C_WaitForSlotEvent with retVal: 00000008
> N: Reader slotnr  0
> N: Reader name  ACS ACR 38U-CCID 0
> N: CbReadersChanged called without a reader description

Frederikus, 2019-01-17:
Hello, to continue the forum discussion here, I'll repost what I posted there:

Windows Server has never been (officially) supported, nor has the viewer or middleware been tested on it.

That being said, I do have an idea where the issue might be comming from.
Having taken a quick look at freerdp, it looks like our windows viewer (with our pkcs11 lib and the windows pcsc library) will talk through freerdp with the linux pcsc deamon.

Now, there are not many differences between our Windows and Linux implementation of pkcs11 (and their usage of pcsc), except for the waitforslotevent function.
There, some linux pcsc deamons used to have issues and the waitforslotevent function was therefore not used in our linux implementation.
This might be what you are running into.

I'll discuss this and if needed run some tests tomorrow with my co-worker who does the linux support to see if we can reproduce this on a pure linux setup, and if we can resolve the issue

nicolasd1980, 2019-02-22:
Hello,
Has there been any movement on this issue as I'm experiencing a similar problem running Server 2008 and connecting to it with linux based thin clients. I'll gladly provide any logs if necessary.

Frederikus, 2019-02-22:
Hello,

We have been testing with activing our waitforslotevent implementation for linux, and the results were looking promessing, but we have not yet tested with a fully ported windows Viewer to Linux.
Which linux kernal version and distro are these clients running?

If you could provide me with the pkcs#11 logs (and the viewer logs as well) from the windows server, I can have a look at it already before my collegue does the linux port.

tdutoit1992, 2019-04-29:
is there enyone with a working cs or vb.net  sln ???

yoe, 2019-09-04:
More information was requested, but never provided. If you do wish to provide the requested information, feel free to reopen the issue. Meanwhile, keeping a bug report open like this serves little purpose.

Additionally, note that the environment in question (Windows Server, Remote Desktop environments) are not actually supported by the official middleware. It may work, and we might accept patches to make it work if they don't break supported configurations, but we don't guarantee that.

closed 2019-09-04 by yoe

tristandegroof, 2020-05-07:
added some extra information to https://github.com/Fedict/eid-mw/issues/86 - hopefully this allows to shed some further light on the issue?