Fedict/eid-mw #75: Support for ChromeOS: missing functionality in APIs

yoe, 2018-06-29:
We tried adding support for ChromeOS, but that can't work, because the ChromeOS APIs are incomplete and don't allow us to add intermediate certificates to the browser. This is strictly required for the Belgian eID

ChromiumOS bug: https://bugs.chromium.org/p/chromium/issues/detail?id=859137&can=2&q=os%3Dchrome

jmeuris, 2019-10-15:
I think there is an alternate approach : using firefox in crostini (linux under Chromeos).

Since Chromeos 77 released the eid readers are visible in the linux container.
tested with Vasco belfius reader :
  jmeuris@penguin:~$ lsusb
  Bus 001 Device 002: ID 1a44:0870 VASCO Data Security International

The issue I have to make it work is that the middleware fails to compile for ARM64 with 'internal compiler errors". And there is no ARM64 precompiled version available.
I will be more than happy to be tester, once a arm64 precompiled version is available.

yoe, 2019-10-16:
> I think there is an alternate approach : using firefox in crostini (linux under Chromeos).

While that might work and I'm happy to take patches to make it possible, it is not an approach that we can officially support. It would, at most, be a workaround.

> Since Chromeos 77 released the eid readers are visible in the linux container.
> tested with Vasco belfius reader :
> jmeuris@penguin:~$ lsusb
> Bus 001 Device 002: ID 1a44:0870 VASCO Data Security [...]

Unfortunately, we can do nothing about that.

"Internal Compiler Error" is th compiler's way of saying it hit a bug in its own code and crashed. That is, it looks like the arm64 compiler that you're using is buggy, and our code just happens to trigger the bug in the compiler.

As said, there is nothing we can do to fix that. What you could do is to try to use a more recent version of the compiler you're using (it might have been fixed). If that doesn't work, please report the bug to the developers of that compiler, with the full output of the bug (not just the "internal compiler error" thing, which is a generic error that doesn't help them).

> And there is no ARM64 precompiled version available.
> I will be more than happy to be tester, once a arm64 precompiled version is available.

There are no plans for arm64 precompiled packages.

We used to have precompiled packages for Debian armhf (i.e., 32-bit ARM with hardware floating point); but these have been dropped since for lack of use.

komasoftware, 2021-01-26:
Something is moving here : https://bugs.chromium.org/p/chromium/issues/detail?id=859137

komasoftware, 2021-01-26:
Star or comment on the issue to encourage Google to make the necessary API changes https://bugs.chromium.org/p/chromium/issues/detail?id=859137

yoe, 2021-01-26:
Yeah, thanks, I missed that. Gave them a bit of feedback on why it's important to us.

LeGrosWinnie, 2024-12-23:
Hello all,

What I don't understand is why this website is working fine and not the other ones (those with CSAM) ?

https://services.e-contract.be/eid-idp/chrome.xhtml