Fedict/eid-mw #76: Missing dependancy «beid-mozilla-extension»

stemy2, 2018-07-17:
Hello,

I can't install eid-mw because the «beid-mozilla-extension» dependancy is missing. Why is this dependancy so important ? After all, i can install it from the Mozilla addons manager. Why is it missing ? And why the help pages don't say anything about it ?

Thanks.

yoe, 2018-07-19:
Yes, it is possible to install the extension manually from the addons
website, but that is just another step which people have to take. If
everything is installed and working once the installation has been
performed, that is much easier. Of course it has to work then, though :-)

What happens when you try to install the beid-mozilla-extension package?
Do you get an error messag? If so, which one? This should just work, so
if there is a problem I'd like to know about it.

The help pages don't say anything about it because this should be a
completely hidden and transparent process under normal circumstances, so
I don't think it is useful to have a FAQ entry about that.

Regards,

yoe, 2018-07-24:
feedback was asked, not received. I also can't reproduce this. Assuming therefore that there isn't an issue, and closing the bug report.

If there is actually an issue, feel free to reopen and provide feedback.

closed 2018-07-24 by yoe