Fedict/eid-mw #78: ../../../libtool: line 1762: g++: command not found

rniesen, 2018-08-28:
Hi,

Ik probeer de code te compileren onder opensuse:tumbleweed:20180502.
De ./configure werkt zonder meldingen.

Bij de make gaat het fout:

CCLD     beid-spr-changepin
  CC       libbeidpkcs11_la-asn1.lo
  CXX      libbeidpkcs11_la-cal.lo
../../../libtool: line 1762: g++: command not found
make[1]: *** [Makefile:1519: libbeidpkcs11_la-cal.lo] Error 1
make[1]: Leaving directory &apos;/tmp/eid-mw/eid-mw/cardcomm/pkcs11/src&apos;
make: *** [Makefile:1856: all-recursive] Error 1

Iemand een idee?

yoe, 2018-08-29:
g++ is the C++ compiler. Since part of the middleware is written in C++, you absolutely need a C++ compiler to compile it. OpenSUSE packages this as [gcc-c++](https://software.opensuse.org/package/gcc-c++), so make sure you have that package installed and try again.

However, the configure script does search for it, so it's weird that you didn't get any errors. Can you please attach your config.log to this issue?

rniesen, 2018-08-29:
Here is the log

gcc is installed.

[config.log](https://github.com/Fedict/eid-mw/files/2331467/config.log)

yoe, 2018-08-29:
Yes, I know gcc is installed. However, gcc-c++ is not. These are not the same. `config.log` shows this, too. It turns out that falling back to g++ in case no C++ compiler was found is [documented behaviour](https://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.69/html_node/C_002b_002b-Compiler.html#C_002b_002b-Compiler), so this is normal.

You need to run `zypper install gcc-c++`. It will work then. Alternatively, after running configure, you can also use the specfile in the `rpm` directory.

Since everything is working as designed, this is not a bug, so closing.

closed 2018-08-29 by yoe