Fedict/eid-mw #79: Build fails on Alpine Linux (musl)

johnlonge, 2018-10-13:
I am trying to build eid-mw on the latest Alpine Linux (3.8.1) but the build fails with musl. Am I doing something wrong?

```
dev@coffee:/tmp/eid-mw-4.4.1-v4.4.1 % make      
make  all-recursive
make[1]: Entering directory '/tmp/eid-mw-4.4.1-v4.4.1'
Making all in cardcomm/pkcs11/src
make[2]: Entering directory '/tmp/eid-mw-4.4.1-v4.4.1/cardcomm/pkcs11/src'
Making all in dialogs/dialogsgtk/po
make[3]: Entering directory '/tmp/eid-mw-4.4.1-v4.4.1/cardcomm/pkcs11/src/dialogs/dialogsgtk/po'
make[3]: Leaving directory '/tmp/eid-mw-4.4.1-v4.4.1/cardcomm/pkcs11/src/dialogs/dialogsgtk/po'
make[3]: Entering directory '/tmp/eid-mw-4.4.1-v4.4.1/cardcomm/pkcs11/src'
  CC       libbeidpkcs11_la-asn1.lo
  CXX      libbeidpkcs11_la-cal.lo
  CC       libbeidpkcs11_la-cert.lo
  CC       libbeidpkcs11_la-encrypt.lo
  CC       libbeidpkcs11_la-general.lo
  CC       libbeidpkcs11_la-pkcs11log.lo
  CC       libbeidpkcs11_la-object.lo
  CC       libbeidpkcs11_la-p11.lo
  CXX      libbeidpkcs11_la-phash.lo
  CC       libbeidpkcs11_la-session.lo
  CC       libbeidpkcs11_la-sign.lo
  CXX      libbeidpkcs11_la-pkcs11util.lo
  CXX      libbeidpkcs11_la-display.lo
  CXX      common/libbeidpkcs11_la-mw_util.lo
  CC       common/libtomcrypt/libbeidpkcs11_la-md5.lo
  CC       common/libtomcrypt/libbeidpkcs11_la-rmd160.lo
  CC       common/libtomcrypt/libbeidpkcs11_la-sha1.lo
  CC       common/libtomcrypt/libbeidpkcs11_la-sha256.lo
  CC       common/libtomcrypt/libbeidpkcs11_la-sha384.lo
  CC       common/libtomcrypt/libbeidpkcs11_la-sha512.lo
  CXX      common/libbeidpkcs11_la-bytearray.lo
  CXX      common/libbeidpkcs11_la-configcommon.lo
  CXX      common/libbeidpkcs11_la-configuration.lo
  CXX      common/libbeidpkcs11_la-datafile.lo
  CXX      common/libbeidpkcs11_la-dynamiclib.lo
  CXX      common/libbeidpkcs11_la-hash.lo
  CXX      common/libbeidpkcs11_la-logbase.lo
common/logbase.cpp: In member function 'bool eIDMW::CLog::open(bool)':
common/logbase.cpp:620:56: error: invalid application of 'sizeof' to incomplete type 'eIDMW::flock'
   m_flock = (struct flock *) malloc(sizeof(struct flock));
                                                        ^
common/logbase.cpp:621:10: error: invalid use of incomplete type 'struct eIDMW::flock'
   m_flock->l_type = F_WRLCK; /* F_RDLCK, F_WRLCK, F_UNLCK    */
          ^~
In file included from common/logbase.cpp:21:0:
common/logbase.h:300:10: note: forward declaration of 'struct eIDMW::flock'
   struct flock *m_flock;
          ^~~~~
common/logbase.cpp:622:10: error: invalid use of incomplete type 'struct eIDMW::flock'
   m_flock->l_whence = SEEK_SET; /* SEEK_SET, SEEK_CUR, SEEK_END */
          ^~
In file included from common/logbase.cpp:21:0:
common/logbase.h:300:10: note: forward declaration of 'struct eIDMW::flock'
   struct flock *m_flock;
          ^~~~~
common/logbase.cpp:623:10: error: invalid use of incomplete type 'struct eIDMW::flock'
   m_flock->l_start = 0; /* Offset from l_whence         */
          ^~
In file included from common/logbase.cpp:21:0:
common/logbase.h:300:10: note: forward declaration of 'struct eIDMW::flock'
   struct flock *m_flock;
          ^~~~~
common/logbase.cpp:624:10: error: invalid use of incomplete type 'struct eIDMW::flock'
   m_flock->l_len = 0; /* length, 0 = to EOF           */
          ^~
In file included from common/logbase.cpp:21:0:
common/logbase.h:300:10: note: forward declaration of 'struct eIDMW::flock'
   struct flock *m_flock;
          ^~~~~
common/logbase.cpp:625:10: error: invalid use of incomplete type 'struct eIDMW::flock'
   m_flock->l_pid = getpid(); /* our PID                      */
          ^~
In file included from common/logbase.cpp:21:0:
common/logbase.h:300:10: note: forward declaration of 'struct eIDMW::flock'
   struct flock *m_flock;
          ^~~~~
common/logbase.cpp: In member function 'void eIDMW::CLog::close()':
common/logbase.cpp:699:10: error: invalid use of incomplete type 'struct eIDMW::flock'
   m_flock->l_type = F_UNLCK; /* tell it to unlock the region */
          ^~
In file included from common/logbase.cpp:21:0:
common/logbase.h:300:10: note: forward declaration of 'struct eIDMW::flock'
   struct flock *m_flock;
          ^~~~~
make[3]: *** [Makefile:1523: common/libbeidpkcs11_la-logbase.lo] Error 1
make[3]: Leaving directory '/tmp/eid-mw-4.4.1-v4.4.1/cardcomm/pkcs11/src'
make[2]: *** [Makefile:1785: all-recursive] Error 1
make[2]: Leaving directory '/tmp/eid-mw-4.4.1-v4.4.1/cardcomm/pkcs11/src'
make[1]: *** [Makefile:482: all-recursive] Error 1
make[1]: Leaving directory '/tmp/eid-mw-4.4.1-v4.4.1'
make: *** [Makefile:410: all] Error 2
```

yoe, 2018-10-14:
Not really, except that the eID middleware requires an flock() implementation, which musl apparently doesn't have. We don't generally support alternative libraries like that; but if you have patches that don't break existing functionality, we'll generally accept them.

Meanwhile though, this is not a problem in eid-mw (at worst it can be a problem in musl), so closing the issue here.

closed 2018-10-14 by yoe

johnlonge, 2018-10-15:
musl-libc does implement struct flock in fcntl.h, but it seems like this was already recently fixed in https://github.com/Fedict/eid-mw/pull/74

Sorry for the noise and thank you for the quick follow-up.

rofl0r, 2018-10-15:
> which musl apparently doesn't have

wrong, musl has it, but *you* didn't include the correct header, and it happened to work on GLIBC due to namespace pollution of another header

> We don't generally support alternative libraries like that

you should reconsider. this point of view means you don't care about portability at all. in this instance merging the tiny fix to make it work on musl actually improved the codebase.

yoe, 2018-10-16:
@rofl0r we have limited time, and can't support everything. Like I said, if someone provides patches that don't break existing functionality we'll generally accept them, but in the absense of that it's not our job to fix the build against every library under the sun.

As it happens, such patches had already been submitted in #74.