Fedict/eid-mw #82: When i output the name, lastname. I am getting accent character broken?

shamun, 2019-01-30:
Hello,

Firstname and Lastname characters such as `ë` is not properly outputing as `ë`, its becoming `..`. What is my mistake?

```
/* ****************************************************************************

* eID Middleware Project.
* Copyright (C) 2011-2012 FedICT.
*
* This is free software; you can redistribute it and/or modify it
* under the terms of the GNU Lesser General Public License version
* 3.0 as published by the Free Software Foundation.
*
* This software is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
* Lesser General Public License for more details.
*
* You should have received a copy of the GNU Lesser General Public
* License along with this software; if not, see
* http://www.gnu.org/licenses/.

**************************************************************************** */

#include "base.h"

/*
* This complete example shows how to read identity data from the card.
*/

CK_RV Beidsdk_GetObjectValue(CK_FUNCTION_LIST_PTR pFunctions, CK_SESSION_HANDLE session_handle, CK_CHAR_PTR pName, CK_VOID_PTR *ppValue, CK_ULONG_PTR pvalueLen);
void Beidsdk_PrintValue(CK_CHAR_PTR pName, CK_BYTE_PTR pValue, CK_ULONG valueLen);
CK_ULONG beidsdk_GetData(void);

int main() {
	CK_ULONG retval = CKR_OK;
	retval = beidsdk_GetData();

	if (retval == CKR_OK) {
		//printf("OK");
	}
	else {
		printf("reader_unplugged");
	}

	//printf("\n");
	//_getch();
}

CK_ULONG beidsdk_GetData()
{
	void *pkcs11Handle;				//handle to the pkcs11 library
	CK_FUNCTION_LIST_PTR pFunctions;		//list of the pkcs11 function pointers
	CK_C_GetFunctionList pC_GetFunctionList;
	CK_RV retVal = CKR_OK;

	//open the pkcs11 library
	pkcs11Handle = dlopen(PKCS11_LIB, RTLD_LAZY);   // RTLD_NOW is slower
	if (pkcs11Handle == NULL) {
		printf("%s not found\n", PKCS11_LIB);
#ifdef WIN32
		DWORD err;
		err = GetLastError();
		printf("err is 0x%.8x\n", err);
		//14001 is "MSVCR80.DLL not found"
#else
		printf("err is %s", dlerror());
#endif
		return CKR_GENERAL_ERROR;
[... 218 lines cut ...]
	retVal = CKR_OK;
	return retVal;
}

void Beidsdk_PrintValue(CK_CHAR_PTR pName, CK_BYTE_PTR pValue, CK_ULONG valueLen) {
	unsigned long counter = 0;
	//printf("%s:\n\n",pName);

	if (pValue != NULL) {
		while (counter < valueLen) {
			if (isprint(*(pValue + counter)))
				printf("%c", *(pValue + counter));
			else
				printf(".", *(pValue + counter));
			counter++;
		}

		printf("#");
	}
	else {
		printf("invalid_insert");
	}

}
```

shamun, 2019-01-30:
On the command prompt when i display accent character, its not broken. 
[image]

But when i am reading and outputting it in command prompt the accent characters are shown as `..`. 
Let me know, what might be the mistake of my code?

Thank you.

Frederikus, 2019-01-30:
These values are stored as UTF8 on the card.

This print function just prints byte per byte, so special chars that are stored over multiple bytes are not printed correct.
I'll have a look at this, thanks for reporting

shamun, 2019-01-30:
I think we should have this? Because otherwise accented output on the command prompt is absolutely broken. 

```
#include <locale.h>
#include <stdio.h>

int main()
{
    int c;
    char* l = setlocale(LC_ALL, "");
    if (l == NULL) {
        printf("Locale not set\n");
    } else {
        printf("Locale set to %s\n", l);
    }
    printf("%ls\n", L"s:\\яшертыHello");
    return 0;
}
```

Please let me know if you have any update. Then i will setup my environment. I lost my entire environment to debug and find the real issue.

Frederikus, 2019-01-31:
Hello,

You could use 
SetConsoleOutputCP(CP_UTF8); to set the windows console to print utf8
And then print the utf8 string that comes from the eid card (keep in mind it is not null terminated).

Frederikus, 2019-02-06:
Also updated our C example

closed 2019-02-06 by Frederikus