Fedict/eid-mw #84: dnd functionality

qlecler, 2019-03-13:
Hello there,

`// Did the user drop a file, or drag data out of the eID-Viewer?
		// NOTE: the necessary changes to the eID-Viewer to
		// support this (without saving a file and dropping that
		// out of a file manager) have not yet been released,
		// and won't be until after tax-on-web.`

[dnd.html](https://github.com/Fedict/eid-viewer/blob/master/xml/dnd.html)

Latest version doesn't seems to have this functionality yet.

Any chance it'll be implemented in the near future?

Thanks 👍

Frederikus, 2019-03-14:
Hello,

There was indeed a bug in the DnD functionality of the eID Viewer, it is fixed in the current beta download of the windows viewer. It has also been fixed in the mac Viewer, but that one hasn't been released yet (to be expected this month).
For the linux Viewer I should check with Wouter, but I'm not aware of any DnD issues there atm.

qlecler, 2019-03-15:
In the meantime, would there be another way ?

I've heard of :

[https://github.com/Trust1Team/t1c-lib-js](https://github.com/Trust1Team/t1c-lib-js)

Frederikus, 2019-03-22:
We've put eID Viewer 4.4.16 for macOS online as future release
(https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer)

This one has had the DnD feature fixed (dragging the photo into e.g. a text editor)

closed 2019-03-22 by Frederikus