Fedict/eid-mw #85: pkcsModule .net example keeps throwing "TOKEN_NOT_RECOGNIZED"

DevEV8, 2019-03-20:
When calling the following function:

`Slot[] slotlist = pkcsModule.GetSlotList(true);` 

Pkcs11 keeps throwing "TOKEN_NOT_RECOGNIZED", I have no idea why it happens, it will work for a while and then it will stop working, and it will throw that.

Frederikus, 2019-03-21:
Hello,

What version of the middleware are you using, and on what OS?
Are you using non-eid cards on the machine as well?
What is the exact setup of your usecase?

Should it be windows, could you try as a test to use the pkcs11 library that comes with the beta version of the eID Viewer (it has some optimizations when it comes to dealing with non-eID cards)

DevEV8, 2019-03-26:
Hi,

I am using windows, and yes the machine has non-eid cards as well.
My middleware version is 4.4.4 build 3838.

Where can I find the beta eId viewer?

My usecase is reading document info from the eID cards and from passports.

Frederikus, 2019-03-26:
Hello,

You can find the beta version of the eID Viewer here:
https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer

DevEV8, 2019-03-26:
Hi,

I downloaded the beta version, copied the beid_ff_pkcs11.dll over to my project, still the same amount of token exceptions.

Frederikus, 2019-03-27:
Hello,

Are you certain that the pkcs11 .NET library is now linked against the new beid_ff_pkcs11.dll?
As in that version,  there is an explicit check in the C_Getslotlist function to not return CKR_TOKEN_NOT_RECOGNIZED (it will return CKR_OK in case of an unrecognized card as it should not care that its not a supported card in the reader at this time).

Just to be sure, can you try to copy and rename the beid_ff_pkcs11.dll to beidpkcs11.dll and overwrite the ones in your project and in your system folders as well?

Wkr,
 Frederik

DevEV8, 2019-04-03:
Hi,

Yes, I have done that, still the same issue, I renamed it in the project code to call that, I also renamed the dll to see if that was the problem.

Is there a project I can build for the new .dll? Maybe that will work.

DevEV8, 2019-04-03:
I have deleted all the .dll's with the old name of the dll, using the new one now, it work's, still throws token exception when it's reading the eid card but that could just be my code trying to check the slot whilst reading.

Thank you.

closed 2019-04-03 by DevEV8