Fedict/eid-mw #86: use of eID Viewer in combination with Igel

pzbdkp, 2019-05-21:
We've discovered an incompatibility between Igel (https://www.igel.com) and the latest eID Viewer version(s). According to Igel, this problem is already know by Fedict.

When using the latest eID Viewer version, it isn't possible to read the eID. The application is aware of eID card insertion (animation in the corner below), but it doesn't read any information.

Igel told us it should work with an older version of the eID Viewer, I've tested with version 4.0.7 build 169. This works, information can be read from the card.

In our organisation we use the xml drag-and-drop functionality of the eID Viewer, which wasn't implemented on this older version.

wesleedb, 2019-05-21:
We have the same problem, hopefully this will be fixed soon!

Frederikus, 2019-05-21:
Hello,

I didn't heard of it untill now, but it seems to be some sort of remote desktop setup with a linux thin client and in your case perhaps a windows server?

This might be the same issue as issue #73 then, caused most lickely by our windows viewer using a virtual card reader or waitforslotevents function that might not be implemented on the thin client.

We do not support remote server setups, but if you provide us with some viewer and pkcs11 logs (https://eid.belgium.be/en/technical-documentation#7483) we can check what causes this, (and fix it if that doesn't mean dropping Windows functionality).

pzbdkp, 2019-05-21:
Igel is some sort of linux distribution which -in our case- connects to a VDI via RDP. 

@Frederikus : Hereby the requested log file:

```
D: log_level combo selected
W: unknown language identifier found in registry:
W: switching language to the system setting
D: CSCblogtest called Handling state transition for event SET_CALLBACKS
D: CSCblogtest called Leaving state LIBOPEN
D: CSCblogtest called Entering state CALLBACKS (target)
D: CSCblogtest called Entering state NO_TOKEN (child)
D: CSCbnewstate called STATE_NO_TOKEN
D: CSCbNewSrc called EID_VWR_SRC_NONE
D: CSCblogtest called Entering state NO_READER (child)
D: CSCbnewstate called STATE_NO_READER
D: CSCblogtest called State transition for SET_CALLBACKS complete
D: CSCblogtest called Handling state transition for event READER_FOUND
D: CSCblogtest called Leaving state NO_READER
D: CSCblogtest called Entering state READY (target)
D: CSCbnewstate called STATE_READY
D: CSCblogtest called State transition for READER_FOUND complete
N: Reader slotnr  0
N: Reader name  Alcor Micro AU9560 00 00
N: Reader slotnr  1
N: Reader name  \\?PnP?\Notification
D: CSCblogtest called Handling state transition for event TOKEN_INSERTED
D: CSCblogtest called Leaving state READY
D: CSCblogtest called Leaving state NO_TOKEN
D: CSCblogtest called Entering state TOKEN (target)
D: CSCbnewstate called STATE_TOKEN
D: CSCbNewSrc called EID_VWR_SRC_CARD
D: CSCblogtest called Entering state TOKEN_ID (child)
D: CSCbnewstate called STATE_TOKEN_ID
```

Frederikus, 2019-05-22:
Any chance you could provide us with the pkcs11 log? (https://eid.belgium.be/en/technical-documentation#7483)
As nothing is wrong with the viewer log (a part from that it stops right before the card content is being read)

pzbdkp, 2019-05-24:
@Frederikus : I've followed your steps in the procedure to the letter. There is no logfile in "c:\programdata\belgium identity card\log".
The log of the eIViewer  is posted in my previous message.

I've tested the eID on this workstation with Firefox; no problem. We also have a in-house developped application with eID. This also works without any problem.
So the problem is related to eID Viewer.

pzbdkp, 2019-05-24:
@Frederikus : Oops. Sorry. I was looking for the log file in "c:\Program Files\Belgium Identity Card\log" instead in "c:\programdata\...". Only after posting here, I've noticed the other path.

The requested p11.log:
```
4296 4596 21.05.2019 17:37:06 C_Initialize()             | I: enter pReserved = 00000000
4296 4596 21.05.2019 17:37:06 C_Initialize()             | S: Initialize this PKCS11 Module
4296 4596 21.05.2019 17:37:06 C_Initialize()             | S: =============================
4296 4596 21.05.2019 17:37:06 C_Initialize()             | I: leave, ret = 0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_lock() acquired
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | S: C_GetSlotList()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[0]: Alcor Micro AU9560 00 00
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=1
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[1]: \\?PnP?\Notification
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_unlock()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: leave, ret = 0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_lock() acquired
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | S: C_GetSlotList()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[0]: Alcor Micro AU9560 00 00
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=1
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[1]: \\?PnP?\Notification
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_unlock()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: leave, ret = 0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_lock() acquired
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | S: C_GetSlotList()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[0]: Alcor Micro AU9560 00 00
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=1
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[1]: \\?PnP?\Notification
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_unlock()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: leave, ret = 336
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_lock() acquired
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | S: C_GetSlotList()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[0]: Alcor Micro AU9560 00 00
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=1
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[1]: \\?PnP?\Notification
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_unlock()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: leave, ret = 0
4296 1032 21.05.2019 17:37:06 C_GetSlotInfo()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotInfo()            | S: C_GetSlotInfo(slot 0)
4296 4596 21.05.2019 17:37:06 C_GetInfo()                | I: enter
4296 4596 21.05.2019 17:37:06 C_GetInfo()                | S: C_GetInfo()
4296 4596 21.05.2019 17:37:06 C_GetInfo()                | I: leave, ret = 0
4296 2268 21.05.2019 17:37:06 C_WaitForSlotEvent(        | I: enter
4296 2268 21.05.2019 17:37:06 C_WaitForSlotEvent(        | S: C_WaitForSlotEvent(flags = 0x0)
4296 1032 21.05.2019 17:37:06 C_GetSlotInfo()            | I: leave, ret = 0
4296 1032 21.05.2019 17:37:06 C_GetSlotInfo()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotInfo()            | S: C_GetSlotInfo(slot 1)
4296 1032 21.05.2019 17:37:06 C_GetSlotInfo()            | I: leave, ret = 0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: enter
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: p11_lock() acquired
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | S: C_GetSlotList()
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: h=0
4296 1032 21.05.2019 17:37:06 C_GetSlotList()            | I: slot[0]: Alcor Micro AU9560 00 00
4368 5788 21.05.2019 17:37:20 C_Initialize()             | I: enter pReserved = 00000000
4368 5788 21.05.2019 17:37:21 C_Initialize()             | S: Initialize this PKCS11 Module
4368 5788 21.05.2019 17:37:21 C_Initialize()             | S: =============================
4368 5788 21.05.2019 17:37:21 C_Initialize()             | I: leave, ret = 0
4368 4432 21.05.2019 17:37:21 C_GetSlotList()            | I: enter
[... 865 lines cut ...]
4368 4432 21.05.2019 17:40:07 C_GetSlotList()            | I: h=0
[... cut ...]
4368 4444 21.05.2019 17:40:07 C_WaitForSlotEvent(        | S: C_WaitForSlotEvent(flags = 0x0)
4368 5788 21.05.2019 17:40:07 C_CloseSession()           | I: enter
4368 5788 21.05.2019 17:40:07 C_CloseSession()           | S: C_CloseSession (session 1)
4368 5788 21.05.2019 17:40:07 C_CloseSession()           | I: leave, ret = 0
4368 5788 21.05.2019 17:40:07 C_GetInfo()                | I: enter
4368 5788 21.05.2019 17:40:07 C_GetInfo()                | S: C_GetInfo()
4368 5788 21.05.2019 17:40:07 C_GetInfo()                | I: leave, ret = 0
```

Frederikus, 2019-05-24:
Thanks for the log

The sequence I read in it is the following:
1) Application 1 (PID 4368) starts and finds 2 card readers
2) 14 seconds later application 2 (PID 3420) starts (During the entire execution of application 2, application 1 will be halted)
3) a few seconds later a card insertion event is detected by application 2, and the card is read out
4) a minute and a half after the card has been read, application 2 presumably ends (stops using pkcs11 in any case)
5) application 1 gets a card event and tries to access the card, but gets an error no card is present (4368 5788 21.05.2019 17:40:07 cal_read_ID_files()        | E: MiddlewareException thrown: 0xe1d00205)
(I presume the card was removed during 4) )
6) No new card insertion event was received, though application 1) is waiting for it
4368 4444 21.05.2019 17:40:07 C_WaitForSlotEvent(        | S: C_WaitForSlotEvent(flags = 0x0)
(But perhaps the card was not inserted again?)

If application 1 is the eID Viewer, this looks to me like either:
-) the linux (igel?) clients pcsc deamon did not send the second card event (card removal)
-) or (more likely) that it was send, but somehow not stored on the server, so that only the card insertion event (as stated in the viewer log) was received after application 2 ended.
And therefore the eID Viewer still shows the "card present" icon.

The main reason that the card wasn't read appears to be that the eID Viewer is blocked for as long as application 2 is active.

Can you try if you can read out the eID Card with the eID Viewer when application 2 is not started?
(Or as a second test by re-entering the eID card after application 2 has ended )

yoe, 2019-09-04:
More information was requested but not provided, so closing the issue. If you do still wish to provide the requested information, feel free to reopen. Meanwhile, this serves no purpose.

Additionally, please note that while it may sometimes work and while we may accept patches, thin client environments are not actually supported by the official middleware.

closed 2019-09-04 by yoe

dewildep, 2020-01-30:
Hello, i do have this issue running 15 new IGEL thin clients. They connect via RDP to a Windows desktop. e-ID viewer is not seeing the card reader in the application. Web applications do not have any issues reading card information from the reader. Is there a way to obtain the old e-ID viewer @pzbdkp mentioned in order to test this? Or will that give us issues with newer e-ID cards?

dewildep, 2020-01-30:
@yoe is it possible to reopen this issue in order to solve this for all Belgium users using linux based thin clients to connect to their environment? I also have direct contact with Igel Support and they are open for modifications to their OS in order to make this work.

yoe, 2020-01-30:
> Hello, i do have this issue running 15 new IGEL thin clients. They connect via RDP to a Windows desktop. e-ID viewer is not seeing the card reader in the application. Web applications do not have [...]

That would be the Java-based viewer at https://github.com/Fedict/eid-viewer, but that one does not support recent cards anymore and is not further developed.

> @yoe is it possible to reopen this issue in order to solve this for all Belgium users using linux based thin clients to connect to their environment? I also have direct contact with Igel Support and they are open for modifications to their OS in order to make this work.

The problem is that we don't have a testing environment, and so we're not entirely sure where the problem lies. If you can figure it out (the eid-viewer is open source, after all), then reasonable patches will most certainly be accepted. However, in the abense of any lead as to what the reason is it doesn't work, there isn't much we can't do.

The fact that it's not supported by us is not the reason it doesn't work; however, it _is_ the reason why we can't fix it.

tristandegroof, 2020-05-07:
[p11.log](https://github.com/Fedict/eid-mw/files/4592622/p11.log)
[viewer.log](https://github.com/Fedict/eid-mw/files/4592623/viewer.log)
[BEID_0.log](https://github.com/Fedict/eid-mw/files/4592625/BEID_0.log)
I created some new logs, which are different from the ones above as there is only one process accessing the card. I annotated the logs where possible with some information as to when a card was inserted and removed. For the viewer log, a card was inserted after the "D: log_level combo selected" statement. After that nothing appears in the viewer log until the card is removed again, this starts with the "N: Reader slotnr  0" entry
What strikes me is the eid viewer in an rds session from igel/linux only seems to "see" the cardreader after a card is inserted, and only lists the slot after the card was removed again.

In contrast, with rds from windows you see it pickup the reader and slot even before a card was inserted (again ,card inserted after the "D: log_level combo selected", rest of the log suppressed for now as I don't think it is relevant?):
D: CSCblogtest called Handling state transition for event SET_CALLBACKS
D: CSCblogtest called Leaving state LIBOPEN
D: CSCblogtest called Entering state CALLBACKS (target)
N: CSCbnewstate called STATE_CALLBACKS
D: CSCblogtest called Entering state NO_TOKEN (child)
N: CSCbnewstate called STATE_NO_READER
N: CSCbNewSrc called EID_VWR_SRC_NONE
D: CSCblogtest called Entering state NO_READER (child)
N: CSCbnewstate called STATE_NO_TOKEN
D: CSCblogtest called State transition for SET_CALLBACKS complete
D: CSCblogtest called eid_vwr_p11_check_reader_list called slotID given = 0
N: CSCblogtest called SLOTEVENT No slot list set, resetting slot list slotID given = 0
D: CSCblogtest called eid_vwr_p11_reset_slot_list resetting slotList
N: CSCblogtest called EVENT_READER_FOUND slotID = 0
D: CSCblogtest called Handling state transition for event READER_FOUND
D: CSCblogtest called Leaving state NO_READER
D: CSCblogtest called Entering state READY (target)
N: CSCbnewstate called STATE_READY
D: CSCblogtest called State transition for READER_FOUND complete
N: Reader slotnr  0
N: Reader name  Broadcom Corp Contacted SmartCard 0
N: Reader slotnr  1
N: Reader name  \\?PnP?\Notification
D: log_level combo selected

On Igel, pcscd is the component providing the access to the slot and card on the thin client, after card insertion that says:
mei 06 18:25:02 RT107-16000945 pcscd[17146]: 99999999 eventhandler.c:423:EHStatusHandlerThread() Card inserted into OMNIKEY CardMan (076B:3022) 3021 00 00
mei 06 18:25:02 RT107-16000945 pcscd[17146]: 00000118 Card ATR: 3B 98 13 40 0A A5 03 01 01 01 AD 13 11
mei 06 18:25:02 RT107-16000945 pcscd[17146]: 00014560 prothandler.c:108:PHSetProtocol() Attempting PTS to T=0
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00271446 ifdwrapper.c:477:IFDControl() Card not transacted: 606
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00000063 ifdwrapper.c:477:IFDControl() Card not transacted: 612
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00011489 ifdwrapper.c:477:IFDControl() Card not transacted: 606
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00000024 ifdwrapper.c:477:IFDControl() Card not transacted: 612
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00004684 ifdwrapper.c:477:IFDControl() Card not transacted: 606
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00000027 ifdwrapper.c:477:IFDControl() Card not transacted: 612
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00016726 ifdwrapper.c:477:IFDControl() Card not transacted: 606
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00000023 ifdwrapper.c:477:IFDControl() Card not transacted: 612
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00218064 ifdwrapper.c:477:IFDControl() Card not transacted: 606
mei 06 18:25:03 RT107-16000945 pcscd[17146]: 00000029 ifdwrapper.c:477:IFDControl() Card not transacted: 612
mei 06 18:26:22 RT107-16000945 pcscd[17146]: 78778297 eventhandler.c:358:EHStatusHandlerThread() Card Removed From OMNIKEY CardMan (076B:3022) 3021 00 00

I understand you say you can't support this situation, but at least some insight into what is goign wrong here would be appreciated... Willing to provide any extra information you need!

Frederikus, 2020-05-07:
Hello Tristan,

What is happening in the eID Viewer is:
One thread (6640) is handling the requests coming from the eID Viewer
application (returning slotlist, slotinfo, reading card data,..)
Another thread (1324) is waiting for pcsc slot changes (when one is
detected, it reports it and starts waiting for slot changes again)

On Windows this works fine.
But what you can see in these logs, is that when using the linux pcscd
deamon, thread 1324 is blocking the execution of pcscd calls made by thread
6640. Each time 1324 detects a card/reader change (and stops waiting for a
little while when reporting the change, 6640 can continue (untill 1324
starts waiting again)).

So the issue is a difference in multithreaded behaviour between the Smart
card service on Windows and the pcscd deamon on linux. Where the smart card
service can handle multi threading from a single pcsc handle, the pcscd
deamon would need a new handle per thread.

In order to add support for the linux pcscd deamon (on windows), our
windows pkcs#11 library should start supporting multiple pcsc handles (to
the same pcsc service/deamon), (or our windows viewer should use multiple
pkcs#11 instances (not nice) ). But these are unwanted changes for an
unsupported usecase.

An easier way to circumvent this issue might be to have a (hidden) option
to run the windows viewer in polling modus (startup parameter, registry
setting,..).
Though this would also mean adding a feature for an unsupported usecase.

tristandegroof, 2020-05-08:
Thanks for the clarification! It sounds like this could also potentially be fixed on the pcscd end? If they implement "multi threading from a single pcsc handle"?
Though that sounds like a big change and thus also unlikely, I could try there.... Unfortunately don't have the skills to propose a fix in code myself.

Frederikus, 2020-05-11:
You're welcome, and thank you for the logs.
I share your view that implementing "multi threading from a single pcsc
handle" in the linux pcscd deamon should indeed resolve this issue.

tristandegroof, 2020-05-13:
to whom it may concern, the answer I received from pcsc-lite:

From: Ludovic Rousseau <ludovic.rousseau@gmail.com>
To: "pcsclite-muscle@lists.infradead.org"
	<pcsclite-muscle@lists.infradead.org>
Subject: Re: [Pcsclite-muscle] multi threading from a single pcsc
	handle / eid-viewer

Hello,

> Hope I am in the right place for the following question. We use igel thin clients (linux based) to rdp to a windows remote desktop farm. Thei gel thin clients use pcsc for their smartcard implementation. On the remote desktop, we have an application eid-viewer, that is used to display the information from a smartcard (electronic identity card or EID) inserted into the thin client. This works well when starting the session from another windows client, but doesn?t work from a linux client using pcsc. According to the eid viewer developpers, the issue is:
> ?What is happening in the eID Viewer is: One thread (6640) is handling the requests coming from the eID Viewer application (returning slotlist, slotinfo, reading card data,..) Another thread (1324) is waiting for pcsc slot changes (when one is detected, it reports it and starts waiting for slot changes again) On Windows this works fine. But what you can see in these logs, is that when using the linux pcscd deamon, thread 1324 is blocking the execution of pcscd calls made by thread 6640. Each time 1324 detects a card/reader change (and stops waiting for a little while when reporting the change, 6640 can continue (untill 1324 starts waiting again)). So the issue is a difference in multithreaded behaviour between the Smart card service on Windows and the pcscd deamon on linux.?
>
> More information and logs can be found in their issue tracker: 
> https://github.com/Fedict/eid-mw/issues/86
>
> Is this ?multi threading from a single pcsc handle? something that could be implemented/fixed in pcsc?

According to pcsc-lite documentation for SCardEstablishContext()
https://pcsclite.apdu.fr/api/group__API.html#gaa1b8970169fd4883a6dc4a8f43f19b67
" Each thread of an application shall use its own SCARDCONTEXT "

So each of the thread in the EID application must establish a different context using SCardEstablishContext().

The developer of the EID application should update it to support pcsc-lite.
Changing this pcsc-lite behavior is a HUGE task. I do not plan to work on it.

Regards,

--
 Dr. Ludovic Rousseau