Fedict/eid-mw #88: DnD no longer working with EID Viewer 4.4.14 and Windows 1903 update (May update)

cliffdeclerck, 2019-08-16:
Since the latest update of Windows 10 (1903) none of the browsers FireFox, Chrome, Edge is accepting the data that is dropped into the browser window.

yoe, 2019-08-19:
Just to confirm, it did work in previous versions of Windows, right?

What exactly goes wrong in 1903? Do you get an error message? If so, which one?

cliffdeclerck, 2019-08-20:
Hi, it did indeed work in previous versions of Windows (<1903).

From that version onwards we get this "prohibition sign" in the cursor when we try to drop in the browser. If we drag and drop it to another application, say Notepadd++, the xml file is still visibile.
Saving the xml file to disk and dropping this file into the browser works as well.

I am guessing there is an extra security measure that requires the binary information that is dropped into the browser having a filename with and extension.

Feel free to contact me for further information.

yoe, 2019-08-22:
Okay, so I tried to reproduce it, but can't.

If I open xml/dnd.html from https://github.com/Fedict/eid-viewer and do a drag-and-drop from eid-viewer v4.4.14 on Windows 1903, everything seems to work just fine for me in Firefox, Chrome, and Edge.

What are you using as the receiving webpage for the drag-and-drop operation?

yoe, 2019-08-22:
(please ignore that message about not being able to reproduce -- I thought I'd updated, but apparently the update got stuck, so that wasn't a 1903)

yoe, 2019-08-22:
Okay, after upgrading to 1903 and installing all updates, I cannot reproduce this issue.

Can you provide more detailed instructions to reproduce?

yoe, 2019-09-04:
More feedback was requested but not received after two weeks, so closing the issue. If you do have more information that allows us to reproduce the issue, feel free to reopen; otherwise, this issue doesn't serve much purpose.

Regards,

cliffdeclerck, 2019-09-04:
Hi, the issue is still open. We however found out that it only occurs on computers with touchscreens.
We also found a solution for this issue and have tested it thoroughly.

Should we send a pull request?

yoe, 2019-09-05:
> Should we send a pull request?

That would be awesome yes, thank you.

MichaelMedi, 2019-09-09:
Could you add me as contributor so I can push the change to a new branch?

Thanks

yoe, 2019-09-09:
Eh, that's not necessary?Fork this repository into your own space on GitHub, push your changes there, then create a push request using the link that will pop up.If you've already cloned this repository and made the changes in that checkout: 'git remote add mine https://GitHub.com/$USER/eid-mw', replacing $USER with your GitHub username, then 'git push mine $BRANCH', replacing $BRANCH by the branch you wish to push. Then, see above.Thanks!

Frederikus, 2019-09-11:
Hello,

Thank you for providing your workaround.
I'll try to get a touchscreen as soon as possible to test the impact of
disabling touchscreen support in the eID Viewer (in regards to scrolling,
resizing,..).

I've read that this issue appears to be fixed on windows insider preview
10.0.18975.1000.
(
https://stackoverflow.com/questions/56354510/why-all-my-wpf-applications-fail-to-drag-outside-of-their-windows-since-windows
)
So we'll be testing that too.

Frederikus, 2019-09-17:
I've tested this with a touchscreen on different Windows 10 versions (and eID Viewer 4.4.14).
As reported, the drag and drop functionality works on version 1709 and 1803, but not on 1903 18362.356
On Windows 10 Insider Preview Build 18362.10019 (19H2) however, drag and drop on the touchscreen works again.
(We've seen reports of other users stating the same (e.g. https://github.com/dotnet/wpf/issues/1323))

So as this is an OS bug (that will be fixed in an upcomming release), we will not be implementing your proposed workaround, but thank you again for bringing this issue to our attention.

We'll keep this bug report open untill an official Windows 10 release has been made that fixes the issue.

Frederikus, 2020-03-26:
Works on Windows 1909, so closing this

closed 2020-03-26 by Frederikus