Fedict/eid-mw #89: EF(ID#Photo) file is cropped after 510 bytes

plazav, 2019-08-28:
I can read EF(ID#RN) file with the code similar to readBinary method in next one example: 

[https://github.com/e-Contract/commons-eid/blob/master/commons-eid-client/src/main/java/be/fedict/commons/eid/client/BeIDCard.java](https://github.com/e-Contract/commons-eid/blob/master/commons-eid-client/src/main/java/be/fedict/commons/eid/client/BeIDCard.java)

But when I am reading image then it is always cropped after around 510 bytes (0xFF twice). 
[image]

Image could be readed with another one application without problems. May be I am missing something?

yoe, 2019-08-28:
For help with developing, please use the [google group](https://groups.google.com/forum/#!forum/eid-middleware-dev) in the future.

An APDU can send up to 512 bytes in one operation, where the last two bytes are the status code. You need to check the status code to ensure that you have read the whole file:

- Status code 0x9000 -> read finished correctly. If the length was 0xFF, then the file is potentially longer; you need to increment the offsets to be certain
- Status code 0x6CXX -> read did not happen, there were less bytes than you asked for. Repeat the same command with "XX" as the new length (Note: some cards do not do this, instead just return less bytes).
- Status code 0x6B00 -> read did not happen, the offset you requested was outside the file.

So, in case you want to read a file of length 300, this is what you need to do:

- READ BINARY (offset 0, length 0xFF)
  - card returns 254 bytes, status code 0x9000
- READ BINARY (offset 254, length 0xFF)
  - card returns 0 bytes, status code 0x6C2E
- READ BINARY (offset 254, length 0x2E)
  - card returns 46 bytes, status code 0x9000

In case of a card just returning less bytes, this is what will happen instead:

- READ BINARY (offset 254, length 0xFF)
  - card returns 46 bytes, status code 0x9000

In case you want to read a file of length 254, the result would instead be:

- READ BINAR (offset 254, length 0xFF)
  - card returns 0 bytes, status code 0x6B00, which tells you that the previous read returned all available bytes

This is not a bug in the code, so closing.

closed 2019-08-28 by yoe

plazav, 2019-08-28:
Thank you, Wouter.
The problem is that I am receiving always 9000 after each read operation. I am removing 2 bytes after each read operation. And if file size is less then 512 then everything is fine. May be after each 512 I should remove 2 bytes additionaly. Will try it. And will ask in google group for the answer.