Fedict/eid-mw #91: Smart Card not recognized

arid88, 2019-09-02:
Hi,

since updating to Win10 it seems my card reader isn't recognised anymore. I tried to install all including drivers and reinstall but still the same issue. eId viewer don't show me any model of reader.  

I've the following codes in the journal:

W: unknown eidviewer_log_level setting identifier found in registry: 
W: Known values are: Error, Warning, Info, Debug
W: switching viewer log level to the default setting (Error)
E: CSCblogtest called eid_vwr_p11_reset_slot_list failed creating new slotList: C_GetSlotList returned = 0X0 number of slots = 0

If someone know how to solve this ?

Frederikus, 2019-09-03:
Hello,

Did you alreadey check with the service desk regarding the unrecognized
card reader?
https://eid.belgium.be/en/contact

They are more aware of issues with certain card readers (drivers) and OS
versions.

Wkr,
 Frederik

yoe, 2019-09-23:
Alternatively, this could also be an issue with the driver of your smartcard reader, i.e., something the manufacturer of your card reader might be able to help you with.

Either way, this is not a bug in the code of the eID middleware, so closing the issue as it won't help.

closed 2019-09-23 by yoe