Fedict/eid-mw #92: symbol lookup error: eid-viewer: undefined symbol: eid_vwr_verify_int_cert

stroobandt, 2019-09-04:
I downloaded and installed the latest Ubuntu versions of `eid-archive`, `eid-viewer` and `eid-mw` from [eid.belgium.be](https://eid.belgium.be/en/linux-eid-software-installation).

Running `/usr/bin/eid-viewer` on Ubuntu 18.04 LTS results in the following error:
`eid-viewer: symbol lookup error: eid-viewer: undefined symbol: eid_vwr_verify_int_cert`

Hee is the output of `about-eid-mw`:

[image]

yoe, 2019-09-04:
Transferred from eid-viewer, since that repository is no longer in use; only the eid-mw one is (the eid-viewer code is under plugins_tools/eid-viewer).

This should only occur if the `eid-viewer` and `libeidviewer0` packages are out of sync. Please try the following:

    dpkg -l eid-viewer libeidviewer0

and check the version numbers. If they're not the same, upgrade the package with the oldest version number.

If the problem reoccurs after that, feel free to reopen this issue -- or contact the service desk, https://eid.belgium.be/en/contact (or your language variant of that page), they will be happy to help you.

closed 2019-09-04 by yoe

stroobandt, 2019-09-04:
$ dpkg -l eid-viewer libeidviewer0
    Desired=Unknown/Install/Remove/Purge/Hold
    | Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
    |/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
    ||/ Name                            Version              Architecture         Description
    +++-===============================-====================-====================-====================================================================
    ii  eid-viewer                      4.4.19v4.4.19-0u1804 amd64                Graphical interface to Belgian Electronic Identity Card
    ii  libeidviewer0:amd64             4.4.19v4.4.19-0u1804 amd64                eid-viewer library

stroobandt, 2019-09-04:
OK, solved it. I found and deleted a couple of older `libeid*` and `libbeid*` files under `/usr/local/lib/`.
Thanks for the hint.