Fedict/eid-mw #95: Mac OSX crash when "Always validate certificate trust"

dietercastel, 2019-10-03:
When enabling the option always validate certificate trust the application crashes on OSX. Doest that mean the certificate can not be validated as well?

[image]
[image]

Frederikus, 2019-10-03:
On which version of OSX are you seeing this?
And with what version of the eID Viewer?

yoe, 2019-10-03:
Also, can you:- click the 'report' button,- click on the text that appears, and press cmd+a to select all of it- press cmd+c to copy it- paste the text into this bug reportThis will give us some more information on where exactly it crashes, which should hopefully allow us to fix it.Thanks,

dietercastel, 2019-10-04:
> On which version of OSX are you seeing this?
> And with what version of the eID Viewer?

Latest version 4.4.16.

Shouldn't integration tests be able to catch such a certificate validation failure?

> Also, can you: - click the 'report' button, - click on the text that appears, and press cmd+a to select all of it - press cmd+c to copy it- paste the text into this bug report This will give us some more information on where exactly it crashes, which should hopefully allow us to fix it.Thanks,
I don't have any reader nearby atm so can't report it rn. I'll get back to you when I can.

yoe, 2019-10-04:
> Shouldn't integration tests be able to catch such a certificate validation failure?

In an ideal world, yes. Unfortunately, we don't have automated tests that detailed yet.

dietercastel, 2019-10-04:
> > Shouldn't integration tests be able to catch such a certificate validation failure?
> 
> In an ideal world, yes. Unfortunately, we don't have automated tests that detailed yet.

Seems like an ideal candidate for a new test then.

yoe, 2019-10-15:
(sorry for the delay, was on holiday last week)

I've been trying to reproduce this, but can't. 4.4.16, 4.4.22 (current beta version) both work just fine with "auto validate" enabled for me, on macOS 10.14 Mojave.

Currently installing the 10.15 update, will try again after that. But please send through the crash report, as requested before, so we can figure out what's happening exactly.

yoe, 2019-10-25:
Didn't manage to reproduce this with Catalina and either of 4.4.16 or 4.4.22.

Please either provide detailed instructions to reproduce, or copy/paste the report to this issue, so that we can investigate what's happening.

Without at least one of those, we cannot investigate what's going wrong. Without being able to reproduce this locally, we have to assume it's a local problem on your side and will act accordingly.

Thanks for your understanding.

yoe, 2019-11-07:
After two more weeks, no further information has been provided, despite our regular request.

Without the requested information (detailed reproduction instructions, or a crash report), it is impossible for us to know what's going wrong, let alone be able to fix it.

In that context, we can only assume this issue is a local configuration problem and not a bug in the eid-viewer, so I'm going to close it.

If you do have the requested information, feel free to reopen.

closed 2019-11-07 by yoe

dietercastel, 2019-11-25:
I managed to get the logs but I couldn't reproduce the crash with 4.4.16 but there is never any validation happening, so I definitely still think that's a bug (even though now it doesn't crash).

```
D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: found data for label ATR
D: found data for label CARD_DATA
D: found data for label carddata_serialnumber
D: found data for label carddata_comp_code
D: found data for label carddata_os_number
D: found data for label carddata_os_version
D: found data for label carddata_soft_mask_number
D: found data for label carddata_soft_mask_version
D: found data for label carddata_appl_version
D: found data for label carddata_glob_os_version
D: found data for label carddata_appl_int_version
D: found data for label carddata_pkcs1_support
D: found data for label carddata_key_exchange_version
D: found data for label carddata_appl_lifecycle
D: found data for label DATA_FILE
D: found data for label card_number
D: converting card_number
D: found data for label chip_number
D: converting chip_number
D: found data for label validity_begin_date
D: found data for label validity_end_date
D: found data for label issuing_municipality
D: found data for label national_number
D: converting national_number
D: found data for label surname
D: found data for label firstnames
D: found data for label first_letter_of_third_given_name
D: found data for label nationality
D: found data for label location_of_birth
D: found data for label date_of_birth
D: converting date_of_birth
D: found data for label gender
D: converting gender
D: found data for label nobility
D: found data for label document_type
D: converting document_type
D: found data for label special_status
D: converting special_status
D: found data for label photo_hash
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_ID
D: found data for label Authentication
D: found data for label Signature
D: found data for label CA
D: found data for label Root
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_CERTS
D: Entering state TOKEN_WAIT (target)
D: Entering state TOKEN_IDLE (child)
D: State transition for READ_READY complete
D: Entering state TOKEN_CERTS (target)
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED
```